pi3d/pi3d_demos #1: OSError: Failed to interpret file '/home/pi/pi3d_demos/textures/earth_clouds.png' as a pickle

stuaxo, 2016-10-18:
I tried to run some of the demos on a pi, with no desktop installed and get a strange error about loading an image as a pickle

```sh
$ python Earth.py 
shader(2) uv_light, b'Compiled'
shader(3) uv_light, b'Compiled'
shader(5) uv_reflect, b'Compiled'
shader(6) uv_reflect, b'Compiled'
shader(8) uv_flat, b'Compiled'
shader(9) uv_flat, b'Compiled'
Traceback (most recent call last):
  File "/home/pi/.virtualenvs/lumen_client/lib/python3.5/site-packages/numpy/lib/npyio.py", line 413, in load
    return pickle.load(fid, **pickle_kwargs)
_pickle.UnpicklingError: A load persistent id instruction was encountered,
but no persistent_load function was specified.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "Earth.py", line 22, in <module>
    cloudimg = pi3d.Texture("textures/earth_clouds.png",True)
  File "/home/pi/pi3d/pi3d/Texture.py", line 125, in __init__
    self.load_disk()
  File "/home/pi/pi3d/pi3d/util/Loadable.py", line 35, in load_disk
    self._load_disk()
  File "/home/pi/pi3d/pi3d/Texture.py", line 207, in _load_disk
    self.image = np.load(self.file_string)['arr_0'] # has to be saved with default key
  File "/home/pi/.virtualenvs/lumen_client/lib/python3.5/site-packages/numpy/lib/npyio.py", line 416, in load
    "Failed to interpret file %s as a pickle" % repr(file))
OSError: Failed to interpret file '/home/pi/pi3d_demos/textures/earth_clouds.png' as a pickle
```

paddywwoof, 2016-10-18:
Hmm, that looks like the latest modification that allow pi3d to still work even if PIL hasn't been installed. In pi3d/constants/__init__.py there is an attempt to load PIL and if it fails the constant PIL_OK is set to False. Subsequently any attempt to load an image resorts to using a compressed numpy array (which seems to use pickle). It all works fine on my RPi with python2 or python3. Are you using a virtual environment, maybe that's relevant?

What happens if you start a python interpreter then
```
>>> from PIL import Image
>>> import pi3d
>>> pi3d.PIL_OK
```

stuaxo, 2016-10-19:
Installing Pillow did indeed fix it.

I had to rebuild python as I'd built it without ncurses-dev being installed the first time, which stopped me using some of the demos.    Then installed libjpeg-dev + Pillow, and everything seems to work now.

closed 2016-10-19 by stuaxo