pi3d/pi3d_demos #2: Some demos run in a letterbox

stuaxo, 2016-10-19:
When I run some demos they appear in a small letterbox:

[image]

(Am running on a pi, without a desktop).

Others seem to run full screen.

paddywwoof, 2016-10-19:
It could be the demos where I used the x,y arguments in Display.create() (that's the case with Amazing.py anyway) This is a (rather cryptic) shortcut interpreted as defining a side and top margin all round the drawing surface. On small screens this can leave a small useful area! Not using x,y,w,h will default full screen.

That looks to be an interesting thing you're doing with multiple screens and IP addresses - is it still under wraps?

stuaxo, 2016-10-19:
Under wraps for now unfortunately :/

Assuming all goes well, then the hardware will be in visible places used by the client though.

That's cool it's a feature, not a bug - will close :)

closed 2016-10-19 by stuaxo