pi3d/pi3d_demos #4: Bounding boxes

spinoza1791, 2019-04-02:
I'd like to use pi3d to create bounding boxes that change shape and location often for my object detector.  I've tried using the pi3d.Lines methods in the loop: 
bbox = pi3d.Lines(vertices=bbox_vertices, material=(1.0,0.8,0.05), closed=True, line_width=4)

But, this way is slow on the Pi (~10FPS).  I found another update method (re_init) using numpy, but it is so confusing, I can't figure out which parts to match to x1, y1, z1 etc.
line_1[:20,:] = np.array([[i*300, i*0, i*20] for i in range(20)])
track_1.buf[0].re_init(line_1)

Is there example code for how to manage a bounding box at a high FPS?

paddywwoof, 2019-04-02:
hi, I'm not sure there is one but I will have a rummage. If there's nothing already I can make one.

It's possible to render cuboids as lines and I think there is a Shape method to find xyz bounds so some combination of those might work.

Is your existing slow code available for me to see?

spinoza1791, 2019-04-03:
Here's a link to my working "slow" code. https://github.com/spinoza1791/coral-pi/blob/master/pi3d_cam.py

I'm extracting bounding box coords from the object detector and then layering the pi3d.Lines over the pi camera preview window, so that the lines create a box around the objects as they move in real-time.  Any thing you can think of to help speed up the real-time box movement would be greatly appreciated! 

I was using Pygame previously, and its very fast (30FPS), but wasn't able to layer it over the picamera preview window.  The Pygame cam video window seemed choppy, so I thought I'd try out pi3d here.

paddywwoof, 2019-04-03:
ok I'm sure it's possible to do this pretty fast, certainly faster than pygame. Quick scan of code, my thoughts: 0. How many rectangles are drawn each frame 1. do you need to open and close your iostream every frame 2. if less than 30 rectangles it's probably fast enough to simply draw the same one scaled and positioned again and again, more than 30 use a single Lines object with a numpy array and re_init. I will give you examples of both to try.

paddywwoof, 2019-04-03:
OK so this is version 1 (I notice that you limit to 5 rectangles so I would expect this to be pretty fast) I will put the re_init() version here too though. I know working with n dimensional arrays and tensors is a bit mind boggling but IMHO it's worth persisting with it as it bumps the performance of python to the level of C in many instances.

```
...
CAMERA = pi3d.Camera(is_3d=False)
font = pi3d.Font("fonts/FreeMono.ttf", font_size=30, color=(0, 255, 0, 255)) # blue green 1.0 alpha
Z = 1.0
bbox_vertices = [[-0.5, 0.5, Z], [0.5, 0.5, Z],
                 [0.5, -0.5, Z], [-0.5, -0.5, Z]] # sides 1.0 x 1.0
bbox = pi3d.Lines(vertices=bbox_vertices, material=(1.0,0.8,0.05), closed=True, line_width=4) 
bbox.set_shader(linshader)
fps = "00.00FPS"
N = 10
fps_txt = pi3d.String(camera=CAMERA, is_3d=False, font=font, string=fps, x=0, y=preview_H/2 - 10, z=1.0)
fps_txt.set_shader(txtshader)
elapsed_ms = 1/1000
ms = "00ms"
ms_txt = pi3d.String(camera=CAMERA, is_3d=False, font=font, string=ms, x=0, y=preview_H/2 - 30, z=1.0)
ms_txt.set_shader(txtshader)
last_tm = time.time()
i = 0

with picamera.PiCamera() as camera:
    camera.resolution = (preview_W, preview_H)
    camera.framerate = max_fps
    rgb = PiRGBArray(camera, size=camera.resolution * 3)
    _, width, height, channels = engine.get_input_tensor_shape()
    camera.start_preview(fullscreen=False, layer=0, window=(preview_mid_X, preview_mid_Y, preview_W, preview_H))
    try:
        stream = io.BytesIO()
        while DISPLAY.loop_running():
            camera.capture(stream, use_video_port=True, format='rgb')
            stream.truncate()
            stream.seek(0)
            input = np.frombuffer(stream.getvalue(), dtype=np.uint8)
            start_ms = time.time()
            results = engine.DetectWithInputTensor(input, top_k=max_obj)
            elapsed_ms = time.time() - start_ms
            ms = str(int(elapsed_ms*1000))+"ms"
            ms_txt.draw()
            ms_txt.quick_change(ms)                
            fps_txt.draw()
            i += 1
            if i > N:
                tm = time.time()
                fps = "{:6.2f}FPS".format(i / (tm - last_tm))
                fps_txt.quick_change(fps)
                i = 0
                last_tm = tm
            if results:
                num_obj = 0
                for obj in results:
                    num_obj = num_obj + 1
                for obj in results:
                    coords = (obj.bounding_box - 0.5) * [[1.0, -1.0]] * mdl_dims # broadcasting will fix the arrays size differences
                    score = round(obj.score,2)
                    (x_scale, y_scale) = coords[1] - coords[0] # will result in 1D array then cast to tuple
                    (x_mid, y_mid) = coords.mean(axis=0) # again will result in tuple
                    bbox.scale(x_scale, y_scale, 1.0)
                    bbox.position(x_mid, y_mid, Z)
                    bbox.draw()

            if keybd.read() == 27:
                break
        stream.close()
    finally:
        keybd.close()
        DISPLAY.destroy()
        camera.close()
```

paddywwoof, 2019-04-03:
Hi, let me know how you get on with these. I've not tested them at all so could well be odd bits of rubbish in there. i.e. don't struggle on trying to get them to work, just let me know what errors are thrown up! NB this re_init() version has to not use GL_LINE_STRIP see the pi3d.Line() line.

```
...
CAMERA = pi3d.Camera(is_3d=False)
font = pi3d.Font("fonts/FreeMono.ttf", font_size=30, color=(0, 255, 0, 255)) # blue green 1.0 alpha
verts = [[0.0, 0.0, 1.0] for i in range(8 * max_obj)] # need a vertex for each end of each side 
bbox = pi3d.Lines(vertices=verts, material=(1.0,0.8,0.05), closed=False, strip=False, line_width=4) 
bbox.set_shader(linshader)
fps = "00.00FPS"
N = 10
fps_txt = pi3d.String(camera=CAMERA, is_3d=False, font=font, string=fps, x=0, y=preview_H/2 - 10, z=1.0)
fps_txt.set_shader(txtshader)
elapsed_ms = 1/1000
ms = "00ms"
ms_txt = pi3d.String(camera=CAMERA, is_3d=False, font=font, string=ms, x=0, y=preview_H/2 - 30, z=1.0)
ms_txt.set_shader(txtshader)
last_tm = time.time()
i = 0

with picamera.PiCamera() as camera:
    camera.resolution = (preview_W, preview_H)
    camera.framerate = max_fps
    rgb = PiRGBArray(camera, size=camera.resolution * 3)
    _, width, height, channels = engine.get_input_tensor_shape()
    camera.start_preview(fullscreen=False, layer=0, window=(preview_mid_X, preview_mid_Y, preview_W, preview_H))
    try:
        while DISPLAY.loop_running():
            stream = io.BytesIO()
            camera.capture(stream, use_video_port=True, format='rgb')
            stream.truncate()
            stream.seek(0)
            input = np.frombuffer(stream.getvalue(), dtype=np.uint8)
            stream.close()
            start_ms = time.time()
            results = engine.DetectWithInputTensor(input, top_k=max_obj)
            elapsed_ms = time.time() - start_ms
            ms = str(int(elapsed_ms*1000))+"ms"
            ms_txt.draw()
            ms_txt.quick_change(ms)                
            fps_txt.draw()
            i += 1
            if i > N:
                tm = time.time()
                fps = "{:6.2f}FPS".format(i / (tm - last_tm))
                fps_txt.quick_change(fps)
                i = 0
                last_tm = tm
            if results:
                num_obj = 0
                for obj in results:
                    num_obj = num_obj + 1
                buf = bbox.buf[0] # alias for brevity below
                buf.array_buffer[:,:3] = 0.0;
                for j, obj in enumerate(results):
                    coords = (obj.bounding_box - 0.5) * [[1.0, -1.0]] * mdl_dims # broadcasting will fix the arrays size differences
                    score = round(obj.score,2)
                    for k in range(8):
                        buf.array_buffer[8 * j + k, 0] = coords[(k + 3) // 4 % 2, 0]
                        buf.array_buffer[8 * j + k, 1] = coords[(k + 1) // 4 % 2, 1]
                buf.re_init(); # 
                bbox.draw() # i.e. one draw for all boxes

            if keybd.read() == 27:
                break
    finally:
        keybd.close()
        DISPLAY.destroy()
        camera.close()
```

spinoza1791, 2019-04-03:
0. How many rectangles are drawn each frame
Due to display contraints, 30 objects is the most I could fit.  The pygame script is set to 20.  I've modified the pi3d (using your numpy code) script to try for 20 now. https://github.com/spinoza1791/coral-pi/blob/master/pi3d-numpy-detector.py.
1. do you need to open and close your iostream every frame 
No, I think that was recommended for memory cleanup, but its fine without that line (no speedup tho).
2. if less than 30 rectangles it's probably fast enough to simply draw the same one scaled and positioned again and again
30 is more than plenty given the display input has to match the model dimensions.
2a. more than 30 use a single Lines object with a numpy array and re_init. 
Using the numpy code vs your non-numpy code yielded the same results ~10FPS

Here's a video comparison, so you can see the bbox "lag" for pygame vs pi3d(numpy):
pygame: https://youtu.be/sv7Nk716wKc (https://github.com/spinoza1791/coral-pi/blob/master/face_pygame.py)
pi3d (numpy version): https://www.youtube.com/watch?v=2khJlgnvU_0

Thanks for your help!

paddywwoof, 2019-04-03:
Hmm that looks not so smooth! I can't see the time being spent either writing to the array, re_initing it or drawing a single buffer - that should easily run 30fps. Is the pygame option basically running the same speed but because it's drawing over the same img that it's used for the face detection there's no 'slippage'.
Before trying different options I think I would do a quick profile (as per some suggestions here http://pi3d.github.io/html/FAQ.html#profiling) Do you know that the face detection will run faster? (You could test that by taking all the display out and just doing the stream capture from the camera, conversion to ndarray and DetectWithInputTensor() in a tight loop, without even doing the profiling)
If you get a passable loop rate from pure detection but the addition of the display process slows it to an unacceptable level then you could run the face detection in another process and feed the results back with a queue. That would make the overlay run smoother (you could even 'tween' the rectangle locations I suppose)
Finally, the broken line system on the pi3d example isn't ideal at the corners, I suppose it could be fixed by offsetting by half the line width (but rather messy) or switch to a continuous line but make it 'invisible' by using the z value to bring it inside the near plane. Or just draw a rectangle multiple times as in the first example I posted.

paddywwoof, 2019-04-03:
Just done a quick experiment drawing N rectangular Line objects with pi3d, onto a full screen transparent background and with N up to 50 it runs flat out at 55 to 60 fps, dropping to around 35fps with 100 object. With the single line numpy re_init method it stays near 60fps up to around 250 rectangles.

spinoza1791, 2019-04-03:
Did some profiling and found:
My original pi3d method: bbox = pi3d.Lines - 11ms
pygame drawing method: 10ms
New pi3d numpy method: buf.re_init();- 1-2ms!
(pi3d numpy script) camera.capture(stream, use_video_port=True, format='bgr'): 65ms
(pygame script) camera.capture(stream, use_video_port=True, format='bgr'): 56ms?
(inference) results = engine.DetectWithInputTensor(input, top_k=max_obj): 30ms (this is an I/O bottleneck of USB2.  USB3 on an OrangePi should get up to 100FPS using the Coral USB Engine)

So, your pi3d numpy method is much faster (x10!).  The way I'm capturing the stream is obviously the bottleneck.  You mentioned a threaded possibility?

paddywwoof, 2019-04-03:
Yes, I'm not sure how it applies to your situation but for some experiments I once did to get camera capture as a pi3d Texture I tried this https://github.com/pi3d/pi3d_demos/blob/master/camera_envsphere_fast.py which basically followed a recipe from the pycamera docs. But there isn't any other animation happening with your system while the image capture and face detection happen so I'm unsure what the gain would be.

It's almost certainly not worth trying to split the image capture and face detection onto different processes as the piping between them would be very slow.

You might be able to use a half resolution image/array for the face detection - that would probably speed it up quite a bit. Even smaller if you can get away with it. Then obviously scale it back up for overlaying on the background video.

I will think a bit about ways round the problem - also about the untidy corners of the rectangles!

paddywwoof, 2019-04-03:
Bit ugly - probably a much simpler way to do this but I think this will work (the first 2 in the formula is half the linewidth)

```
                    score = round(obj.score,2)
                    for k in range(8):
                        buf.array_buffer[8 * j + k, 0] = (coords[(k + 3) // 4 % 2, 0] +
                            2 * ((k // 4) % 2 * 2 - 1) * ((k // 2) % 2))
                        buf.array_buffer[8 * j + k, 1] = (coords[(k + 1) // 4 % 2, 1] +
                            2 * (((k + 2) // 4) % 2 * 2 - 1) * (((k + 2) // 2) % 2))
                buf.re_init(); # 
```

paddywwoof, 2019-04-03:
Actually something like this would be much simpler

```
X_OFF = [0, 0, -1, -1, 0, 0, 1, 1]
Y_OFF = [-1, -1, 0, 0, 1, 1, 0, 0]
...
                        buf.array_buffer[8 * j + k, 0] = coords[(k + 3) // 4 % 2, 0] + 2 * X_OFF[k]
                        buf.array_buffer[8 * j + k, 1] = coords[(k + 1) // 4 % 2, 1] + 2 * Y_OFF[k]
```

spinoza1791, 2019-04-04:
Thank you, I've integrated this last change into: https://github.com/spinoza1791/coral-pi/edit/master/pi3d-numpy-detector-dev.py for profiling.
Results:
1 bbox (my face) ~ 2ms
16 bbox (faces) ~ 11ms  
Every bit makes a huge difference once the number of bboxes gets larger!  I honestly think the "lag" is largely that using picamera preview (GPU) is a much higher frame rate of display than the pygame display (CPU) and so the pi3d boxes appear to lag behind the faster frames, but if pygame were having to chase this higher GPU frame rate, I believe it would appear to lag just as much.  I think if we can find a faster way to do "camera.capture", (65ms -> 10ms), then I could eventually take advantage of the 100FPS potential of USB3, without bbox lag.  Currently, on RPi with USB2, the limit is 30FPS, so getting the camera.capture down to 30ms would be good enough.

paddywwoof, 2019-04-04:
Hmm. I wasn't sure where the USB comes in but just looked at your github intro and see you say to plug the accelerator into USB so is that all handled by the edgetpu.detection.engine? I would have thought USB2 could pipe 30MB/s which should be a reasonable number of frames (I noticed that you're using 320x320 pixels which is already rather reduced - do you start missing faces if you reduce it further, say to 240?) I think I found reducing the image size was the most significant thing when I messed around with the camera. What happens if you specify resize argument in camea.capture()?

Also will the detector work in greyscale - that would reduce the volume going to the detector by a factor of 3. I can't see anything in the docs about that, just says 'input_tensor'. Easy enough to try, you can convert an rgb numpy array shape ``A.shape => (H,W,3)`` to greyscale siimply by ``(A * [0.2989, 0.5870, 0.1140]).sum(axis=2).astype(np.uint8)``

spinoza1791, 2019-04-08:
Did my best to integrate my latest script with the threading example from camera_envsphere_fast.py here:  https://github.com/spinoza1791/coral-pi/blob/master/fast-pi3d-cam.py
FPS looks better, but the bbox's seem to overload the pipe and "stack up" without clearing the buffer fast enough.  This lag and refresh issue seems especially pronounced with multiple objects (I'll create a video link later today, so you can see).  Can you see anything inherently wrong with fast-pi3d-cam.py to explain the pile up of boxes in the preview window?  Perhaps there's a better way to marry up the refresh of the boxes with the frames, so that they are in sync?  Thanks again!

paddywwoof, 2019-04-08:
Hi, good that it seems to have improved FPS a bit. I'll look at the code in more detail later but I can see that there might be an issue in having four instances of ImageProcessor but only one result global. Also each instance creates a new DetectionEngine which might might work but might not actually increase the detection capability, especially if it has to go through the same USB, and there could be complications created. I see that the original camera_envsphere_fast overwrites the same numpy array passed as a global but that might work because pretty much all the time goes in the bytesio streaming. For your situation a big part of the time will be spent on the line ``self.output = self.engine.DetectWithInputTensor(self.input_val, top_k=4)``

I'm not sure what you mean by 'stack up', may be clearer  after seeing the video. Is it that the DetectionEngine can't process the stream as fast as the images are being written or does the number of rectangles in the results array get bigger and bigger?

There might need to be some additional flagging. In the original I only 'remake' the texture when the ``new_pic`` gets set and I unset it as soon as the array has been used. In your code that would be the equivalent of something like

```
...
    if new_pic:  
      if results: # != None and empty_results == 1:
        num_obj = 0
        for obj in results:
          num_obj = num_obj + 1   
          buf = bbox.buf[0] # alias for brevity below
          buf.array_buffer[:,:3] = 0.0;
          for j, obj in enumerate(results):
            coords = (obj.bounding_box - 0.5) * [[1.0, -1.0]] * mdl_dims
            score = round(obj.score,2)
            ix = 8 * j
            buf.array_buffer[ix:(ix + 8), 0] = coords[X_IX, 0] + 2 * X_OFF
            buf.array_buffer[ix:(ix + 8), 1] = coords[Y_IX, 1] + 2 * Y_OFF
          buf.re_init()
          new_pic = False 
        bbox.draw() # i.e. one draw for all boxes
...
```

but you might need a list of flags and results, one for each threaded detector

spinoza1791, 2019-04-24:
Success! I played around with the script and was able to get it smooth here: https://github.com/spinoza1791/coral-pi/blob/master/pi3d-bbox-detection.py

However, I also have need for the labels (i.e., "face") to appear as text just over the bboxes as they move, which I assume means the pi3d.String will need an array and re_init() somehow.  Do you know how this can be done with the String class here?

closed 2020-09-18 by paddywwoof