pi3d/pi3d_demos #6: How to exit from PictureFrame.py  one time that I launched it?

matteoraggi, 2019-05-27:
Hi, from raspbian stretch, I launch this from the GUI and it works perfect! 
But then to exit from it the only way is to plug off the electricity to the raspberry.
Is there a more intelligent way to exit from python3 PictureFrame.py one time launched?

paddywwoof, 2019-05-27:
Matteo, did you try ESC? What happened?

matteoraggi, 2019-05-27:
Nothing it continue to play as if nothing new happened..

paddywwoof, 2019-05-28:
Matteo, what happens with the other demos, for instance Minimal or Blur? I think they all use the same keyboard reading functionality. It might be worth putting a line into PictureFrame to make it break and see what happens (around line 112)
```
    pic_num = (pic_num + 1) % nFi # wraps to start
    if pic_num > 5:
        break
    if (pic_num % CHKNUM) == 0: # this will shuffle as well
```

closed 2020-09-18 by paddywwoof