pi3d/pi3d_demos #10: make blurred Background/Edges on Pictureframe.py

Dud3z, 2019-11-14:
Hi i would really like to see how to get blurry edges if the aspect ration of the picture is not right.

Like This Picture: [Blurry Edges Example](https://imgur.com/hbgn0SE)

As of right now i did not find a way to get this.

paddywwoof, 2019-11-14:
OK this has cropped up a few times so I will make a version of the PictureFrame demos that does this. Might be a couple of days.
Paddy

Dud3z, 2019-11-15:
you are AWESOME!

paddywwoof, 2019-11-16:
alpha of script here https://github.com/pi3d/pi3d_demos/blob/develop/PictureFrame_blur.py I haven't tested it on RPi etc yet. If you get a chance to try it let me know how it runs.

Dud3z, 2019-11-18:
seems to work quite good on my pi3 b+. it would be very nice to control the blur and the crop.

paddywwoof, 2019-11-18:
Hi. Couple of tweaks added yesterday. Can make blur amount a variable easily enough but what do you mean by controlling cropping?

Dud3z, 2019-11-19:
i meant the zoom factor of the image on the side/background.

paddywwoof, 2019-11-19:
OK. I'm not sure if it will really work with zoom factors that result in background (blurred images) smaller than the screen width or height but I will look at making the factor a user settable const going from 1.0 which will match the width or height to whatever. I will let you know when I've pushed the latest changes to github develop branch

paddywwoof, 2019-11-19:
new version up now. Let me know what you think. I  will aim to include this in the normal PictureFrame as an option and put it up in pi3d/pi3d_demos master branch

```
SHOW_NAMES = False
CHECK_DIR_TM = 60.0 # seconds to wait between checking if directory has changed
#####################################################
BLUR_EDGES = True # use blurred version of image to fill edges - no point if FIT = False
BLUR_AMOUNT = 12 # larger values than 12 will increase processing load quite a bit
BLUR_ZOOM = 1.0 # must be >= 1.0 which expands the backgorund to just fill the space around the image
#####################################################
```

Dud3z, 2019-11-19:
this is perfect! it does exactly what i tought it should :) you made my day! my blur_amount is at 48 and it takes only a second longer to load up the script!

klaumsco, 2019-12-02:
I've added an alpha to the blurred image in the code below:
`def tex_load(fname, orientation, size=None):
  try:
    im = Image.open(fname).convert("RGB")
    if orientation == 2:
        im = im.transpose(Image.FLIP_LEFT_RIGHT)
    if orientation == 3:
        im = im.transpose(Image.ROTATE_180) # rotations are clockwise
    if orientation == 4:
        im = im.transpose(Image.FLIP_TOP_BOTTOM)
    if orientation == 5:
        im = im.transpose(Image.FLIP_LEFT_RIGHT).transpose(Image.ROTATE_270)
    if orientation == 6:
        im = im.transpose(Image.ROTATE_270)
    if orientation == 7:
        im = im.transpose(Image.FLIP_LEFT_RIGHT).transpose(Image.ROTATE_90)
    if orientation == 8:
        im = im.transpose(Image.ROTATE_90)
    if BLUR_EDGES and size is not None:
      wh_rat = (size[0] * im.size[1]) / (size[1] * im.size[0])
      if abs(wh_rat - 1.0) > 0.01: # make a blurred background
        (sc_b, sc_f) = (size[1] / im.size[1], size[0] / im.size[0])
        if wh_rat > 1.0:
          (sc_b, sc_f) = (sc_f, sc_b) # swap round
        (w, h) =  (round(size[0] / sc_b / BLUR_ZOOM), round(size[1] / sc_b / BLUR_ZOOM))
        (x, y) = (round(0.5 * (im.size[0] - w)), round(0.5 * (im.size[1] - h)))
        box = (x, y, x + w, y + h)
        blr_sz = (int(x * 512 / size[0]) for x in size)
        im_b = im.resize(size, resample=0, box=box).resize(blr_sz)
        im_b = im_b.filter(ImageFilter.GaussianBlur(BLUR_AMOUNT))
        im_b.putalpha(round(255*EDGE_ALPHA)) # to apply the same EDGE_ALPHA as the no blur method.
        im_c = im_b
        im_c = im_c.resize(size, resample=Image.BICUBIC)
        im = im.resize((int(x * sc_f) for x in im.size), resample=Image.BICUBIC)
        im_c.paste(im, box=(round(0.5 * (im_c.size[0] - im.size[0])),round(0.5 * (im_c.size[1] - im.size[1]))))
        im = im_c # have to do this as paste applies in place
    tex = pi3d.Texture(im, blend=True, m_repeat=True, automatic_resize=True)
  except Exception as e:
    print('''Couldn't load file {} giving error: {}'''.format(fname, e))
    tex = None
  return tex
`

This applies the same alpha % to the background ask the older EDGE_ALPHA did. 

Scott

klaumsco, 2019-12-02:
I've added a alpha to the blurred edges with the snippet tex_load below:

`
def tex_load(fname, orientation, size=None):
  try:
    im = Image.open(fname).convert("RGB")
    if orientation == 2:
        im = im.transpose(Image.FLIP_LEFT_RIGHT)
    if orientation == 3:
        im = im.transpose(Image.ROTATE_180) # rotations are clockwise
    if orientation == 4:
        im = im.transpose(Image.FLIP_TOP_BOTTOM)
    if orientation == 5:
        im = im.transpose(Image.FLIP_LEFT_RIGHT).transpose(Image.ROTATE_270)
    if orientation == 6:
        im = im.transpose(Image.ROTATE_270)
    if orientation == 7:
        im = im.transpose(Image.FLIP_LEFT_RIGHT).transpose(Image.ROTATE_90)
    if orientation == 8:
        im = im.transpose(Image.ROTATE_90)
    if BLUR_EDGES and size is not None:
      wh_rat = (size[0] * im.size[1]) / (size[1] * im.size[0])
      if abs(wh_rat - 1.0) > 0.01: # make a blurred background
        (sc_b, sc_f) = (size[1] / im.size[1], size[0] / im.size[0])
        if wh_rat > 1.0:
          (sc_b, sc_f) = (sc_f, sc_b) # swap round
        (w, h) =  (round(size[0] / sc_b / BLUR_ZOOM), round(size[1] / sc_b / BLUR_ZOOM))
        (x, y) = (round(0.5 * (im.size[0] - w)), round(0.5 * (im.size[1] - h)))
        box = (x, y, x + w, y + h)
        blr_sz = (int(x * 512 / size[0]) for x in size)
        im_b = im.resize(size, resample=0, box=box).resize(blr_sz)
        im_b = im_b.filter(ImageFilter.GaussianBlur(BLUR_AMOUNT))
        im_b.putalpha(round(255*EDGE_ALPHA)) # to apply the same EDGE_ALPHA as the no blur method.
        im_c = im_b
        im_c = im_c.resize(size, resample=Image.BICUBIC)
        im = im.resize((int(x * sc_f) for x in im.size), resample=Image.BICUBIC)
        im_c.paste(im, box=(round(0.5 * (im_c.size[0] - im.size[0])),round(0.5 * (im_c.size[1] - im.size[1]))))
        im = im_c # have to do this as paste applies in place
    tex = pi3d.Texture(im, blend=True, m_repeat=True, automatic_resize=True)
  except Exception as e:
    print('''Couldn't load file {} giving error: {}'''.format(fname, e))
    tex = None
  return tex

`

It applies the same alpha as the old EDGE_ALPHA method

Scott

klaumsco, 2019-12-02:
I've added a alpha to the blur with the tex_load below:

```
def tex_load(fname, orientation, size=None):
  try:
    im = Image.open(fname).convert("RGB")
    if orientation == 2:
        im = im.transpose(Image.FLIP_LEFT_RIGHT)
    if orientation == 3:
        im = im.transpose(Image.ROTATE_180) # rotations are clockwise
    if orientation == 4:
        im = im.transpose(Image.FLIP_TOP_BOTTOM)
    if orientation == 5:
        im = im.transpose(Image.FLIP_LEFT_RIGHT).transpose(Image.ROTATE_270)
    if orientation == 6:
        im = im.transpose(Image.ROTATE_270)
    if orientation == 7:
        im = im.transpose(Image.FLIP_LEFT_RIGHT).transpose(Image.ROTATE_90)
    if orientation == 8:
        im = im.transpose(Image.ROTATE_90)
    if BLUR_EDGES and size is not None:
      wh_rat = (size[0] * im.size[1]) / (size[1] * im.size[0])
      if abs(wh_rat - 1.0) > 0.01: # make a blurred background
        (sc_b, sc_f) = (size[1] / im.size[1], size[0] / im.size[0])
        if wh_rat > 1.0:
          (sc_b, sc_f) = (sc_f, sc_b) # swap round
        (w, h) =  (round(size[0] / sc_b / BLUR_ZOOM), round(size[1] / sc_b / BLUR_ZOOM))
        (x, y) = (round(0.5 * (im.size[0] - w)), round(0.5 * (im.size[1] - h)))
        box = (x, y, x + w, y + h)
        blr_sz = (int(x * 512 / size[0]) for x in size)
        im_b = im.resize(size, resample=0, box=box).resize(blr_sz)
        im_b = im_b.filter(ImageFilter.GaussianBlur(BLUR_AMOUNT))
        im_b.putalpha(round(255*EDGE_ALPHA)) # to apply the same EDGE_ALPHA as the no blur method.
        im_c = im_b
        im_c = im_c.resize(size, resample=Image.BICUBIC)
        im = im.resize((int(x * sc_f) for x in im.size), resample=Image.BICUBIC)
        im_c.paste(im, box=(round(0.5 * (im_c.size[0] - im.size[0])),round(0.5 * (im_c.size[1] - im.size[1]))))
        im = im_c # have to do this as paste applies in place
    tex = pi3d.Texture(im, blend=True, m_repeat=True, automatic_resize=True)
  except Exception as e:
    print('''Couldn't load file {} giving error: {}'''.format(fname, e))
    tex = None
  return tex
```

This applies the same alpha as the old EDGE_ALPHA method.

Scott

paddywwoof, 2019-12-02:
Scott, thanks for that. It makes sense to leave the extra flexibility in so I will change the code accordingly. I think I assumed, slightly illogically, that people would either want the blurred version OR the mirrored version with variable transparency!

paddywwoof, 2019-12-02:
Scott, thanks for that. It makes sense to leave the extra flexibility in so I will change the code accordingly. I think I assumed, slightly illogically, that people would either want the blurred version OR the mirrored version with variable transparency!

```
 def tex_load(fname, orientation, size=None):
   try:
-    im = Image.open(fname).convert("RGB")
+    im = Image.open(fname)
+    im.putalpha(255) # this will convert to RGBA and set alpha to opaque
     if orientation == 2:
         im = im.transpose(Image.FLIP_LEFT_RIGHT)
     if orientation == 3:
@@ -90,8 +91,9 @@ def tex_load(fname, orientation, size=None):
         box = (x, y, x + w, y + h)
         blr_sz = (int(x * 512 / size[0]) for x in size)
         im_b = im.resize(size, resample=0, box=box).resize(blr_sz)
-        im_b = im_b.filter(ImageFilter.BoxBlur(BLUR_AMOUNT))
+        im_b = im_b.filter(ImageFilter.GaussianBlur(BLUR_AMOUNT))
         im_b = im_b.resize(size, resample=Image.BICUBIC)
+        im_b.putalpha(round(255 * EDGE_ALPHA))  # to apply the same EDGE_ALPHA as the no blur method.
         im = im.resize((int(x * sc_f) for x in im.size), resample=Image.BICUBIC)
```

klaumsco, 2019-12-02:
All I implemented was the alpha from the edge mirror code. That is likely what folks will want to be able to control, since the blur at 100% opacity stands out a little too much for my liking.   
  

  
Thanks again for all your work on this!
  

  
  
  
 Scott Klauminzer

closed 2020-09-18 by paddywwoof