pi3d/pi3d_demos #13: PictureFrame.py not working anymore

nicolasborn, 2020-01-05:
Thank you @paddywwoof for the bugfix in commit d64fe82981bdabbe805492924a8e05e5747d1325. i was searching quite a while why it didn't run in autostart. but your newest version isn't working anymore on my raspberry pi 3 with raspbian buster. Since I'm not using Ken Burns, it must have to do with one of the following changes:
`im = Image.open(fname)
im.putalpha(255) # this will convert to RGBA and set alpha to opaque`
or
`im_b = im_b.filter(ImageFilter.GaussianBlur(BLUR_AMOUNT))
im_b = im_b.resize(size, resample=Image.BICUBIC)
im_b.putalpha(round(255 * EDGE_ALPHA))  # to apply the same EDGE_ALPHA as the no blur method.`
with the latest version, the screen stays black and no image is displayed.

paddywwoof, 2020-01-05:
@nicolasborn sorry if I've broken something. I'll have a look and see if I can find the problem. And fix it. Which of the options do to have set? KEYBOARD, KENBURNS False BLUR_EDGES True?

paddywwoof, 2020-01-05:
@nicolasborn does it only not work with autostart? Have you tried running from a terminal? The only thing that stands out as being different is the ``Texture(...automatic_resize=False...)`` Which might have an impact if you are using very large images and not setting BLUR_EDGES True. Does it work ok with the default PIC_DIR ``/home/pi/pi3d_demos/textures``?

nicolasborn, 2020-01-06:
@paddywwoof I will try it in the evening with smaller pictures (the pictures are around 2-3mb). it also doesn't work when running from a terminal. i'm not sure how i've set BLUR_EDGES but I think I haven't changed that so it should be in the default setting.

paddywwoof, 2020-01-06:
@nicolasborn I'm not really sure why the resizing to standard widths causes issues (if that is the cause) but it might be a problem if they are too wide or high.
The first thing I would try would be [the original setup unchanged](https://github.com/pi3d/pi3d_demos/blob/master/PictureFrame.py#L26) (you have to stop the program with AltF4 if you start from command line as KEYBOARD is disabled).
If that works then maybe try your image directory but make the resize happen, you could just change the False to True but in the actual code I will do something like:
```python
        im_b.paste(im, box=(round(0.5 * (im_b.size[0] - im.size[0])),
                            round(0.5 * (im_b.size[1] - im.size[1]))))
        im = im_b # have to do this as paste applies in place
      do_resize = False
    else:
      do_resize = True
    tex = pi3d.Texture(im, blend=True, m_repeat=True, automatic_resize=do_resize, free_after_load=True)
  except Exception as e:
    print('''Couldn't load file {} giving error: {}'''.format(fname, e))
```
i.e. don't do the resize if the edges are being blurred (as there's already resizing going on to fit the screen).
Let me know how you get on.

Paddy

PS what are you running this on? (board/os/GPU driver etc)

nicolasborn, 2020-01-06:
@paddywwoof raspberry pi 3 with raspbian buster.  A8 gl driver set to g1 legacy.
will try it with the original setup in the evening.

paddywwoof, 2020-01-06:
@nicolasborn Hi, I've reproduced your error now (by increasing some images to 5120x5120) and patched the file https://github.com/pi3d/pi3d_demos/commit/5b573d6648b3d1b1ece3c604afe631d02a81c792

Thanks for spotting this and reporting it so quickly!

Paddy

nicolasborn, 2020-01-06:
@paddywwoof it works! thank your very much, paddy!
greetings, nicolas

closed 2020-01-06 by nicolasborn

nicolasborn, 2020-01-06:
@paddywwoof I still get some errors but I think I had them before your change. but do you know why the exif data of my images cannot be read?

trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 36867
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 274
trying to read exif 'NoneType' object is not subscriptable
trying to read exif 274
...

paddywwoof, 2020-01-06:
I'm not sure exactly, but the version of python-pillow that ships with raspbian buster is quite a way behind the current one and quite a few patches are on the exif processing. It might be worth trying to upgrade it with
```bash
pip3 install Pillow --upgrade --user
```