pi3d/pi3d_demos #14: Pictureframe.py randomly fail to show pictures

radrad2, 2020-01-12:
Hi guy,
I just love the project, it is very nice when it works good.
I install it on raspberry pi 3 and have a weird issue, every few images, one of the images failed to load and all I see is black screen until the next one. 
It is now fixed pictures or time, just very random. 
sometimes it is one of 5 images and other time 2 out of 3 which is too much..
image size is around 5 MB if it helps.
let me know if you need any extra information

paddywwoof, 2020-01-12:
Hi, there was this issue https://github.com/pi3d/pi3d_demos/issues/13 recently where one of the upgrades had caused an issue. If you set the option on line 44 or so
BLUR_EDGES = True
and it works ok then that could be the cause of the problem. It's the width of the images that's to blame.
If that does fix it then you can get the latest version from this github repo.
If that doesn't fix it then I will probably need some more info: when you downloaded the repo, the w and h of pictures that show v. those that don''t etc.

Thanks for you help.

radrad2, 2020-01-12:
Thank you for the super quick answer.
I'm using the latest version (downloaded again today to be sure)
same behavior with original config except for the pictures path.
I tried to change the value for BLUR_EDGES but no change...

if it helps, when I run the script i see in the terminal 
b'GLES2' 
glGetError 0x500

example details of images:
1. h: 5120 w: 3840   size: 3.4MB
2. h: 3120 w: 4160   size: 3.3MB
3. h: 3648 w: 5472   size 5.4MB
All are JPG type. 
The weirdest thing is that sometimes It does show them successfully and sometimes not .

paddywwoof, 2020-01-12:
Oh well, it would have been nice if there was a simple explanation! I suppose those images are quite large for the relatively low capacity RPi. Image #3 will expand to 80MB but the resizing should happen pretty much as the images load, and certainly before allocating GPU memory. What GPU memory split have you set? I'm not sure how it's relevant (apart for the BLUR_EDGES) but what are the dimensions of the screen?

Also, might it be related to images that are higher than wide - say the image that follow a tall image? I will try creating some large images with the dimensions you listed (and others) and see if I can reproduce your fault. If I can then I will try to pin down the point where it stops working.

paddywwoof, 2020-01-13:
@radrad2 Hi, I think I've managed to reproduced your error (famous last words not withstanding):

0. Running from the command line without Xserver started.
1. I put five files in a directory and resized them to 3840x5120.jpg, 4160x3120.jpg, 5472x3648.jpg, 5472x3648.png and 5120x5120.jpg
2. I set PictureFrame KEYBOARD and SHOW_NAMES to True (just for debugging purposes) and set the picture dir to 1. above.
3. in `sudo raspi-config` I set `Advanced Options/GL driver` to legacy and `Memory split` to 64 (then rebooted)
4. `$ cd ~/pi3d_demos`, `$ python3 PictureFrame.py` -> occasional missing images.

If I change the GL driver option to Fake KMS and start with
`$ sudo xinit /usr/bin/python3 /home/pi/pi3d_demos/PictureFrame :0`
Then it seems to run ok with 64 GPU memory. If I increased the GPU memory to 128 then it ran using the legacy driver.

Could you let me know the settings you were using in raspi-config and whether you were running with or without the Xserver started.

Thanks again for your help with this.

Paddy

bukem, 2020-06-18:
Hi guys,

I had the same issue and increasing GPU memory from default 64MB to 128MB really fixed the problem.

@paddywwoof , thanks!

closed 2020-12-13 by paddywwoof