pi3d/pi3d_demos #16: exif issues with raspberry pi zero w

sl4m01, 2020-02-29:
Hi,
i do have 2 issue with showing the slideshow on a raspberry pi zero w:
1. after starting the script i am getting a lot of this warning:
trying to read exif 274
...
trying to read exif 'NoneType' object is not subscriptable
...
trying to read exif 36867
...
glGetError 0x500
2. Sometimes the screen just shows black, maybe some images can not be display (the are all jpg and ~0.3-4mb big).

Can someone help? If needed i can copy my config here.

paddywwoof, 2020-02-29:
@sl4m01 thanks for feedback. Do you know if the image files have valid exif info? The exif handling on the version of PIL that's shipped with raspbian buster is a bit flaky and has lots of patches in the current form. This is bit unfortunate as debian (which dictates behavior to raspbian) will stick with Pillow (aka PIL) v. 5.4.1 until the middle of 2021. Meanwhile Pillow stable is already at v 7.0.0.

So the first thing I would suggest is upgrading that to the latest version:
`pip3 install Pillow --upgrade`
(you might need an additional --user on the end, or (ideally not) start with sudo)

The issue with occasionally black images is probably due to image size. This has cropped up and been discussed in issues on this forum - have to tried the suggestions there?

sl4m01, 2020-02-29:
Thank you a lot for your fast answer.
To be honest, i dont know if the pictures do have a valid exif info. They are from different sources collected over the last 10 years (smart phones, cameras). I do sync them from dropbox to raspberry pi. Can you tell me how to check?
I already performed pip3 install Pillow --upgrade. At least it installed something (i am a raspberry pi noob). However no change.

I tried the hints in #14 
BLUR_EDGES = True is already set.
Changing to Full or Fake KMS i receive this error if i want to run Pictureframe.py:
`Traceback (most recent call last):
  File "PictureFrame.py", line 254, in <module>
    display_config=pi3d.DISPLAY_CONFIG_HIDE_CURSOR, background=BACKGROUND)
  File "/usr/local/lib/python3.7/dist-packages/pi3d/Display.py", line 563, in create
    display_config=display_config, window_title=window_title, use_glx=use_glx)
  File "/usr/local/lib/python3.7/dist-packages/pi3d/util/DisplayOpenGL.py", line 102, in create_display
    assert self.context != EGL_NO_CONTEXT and self.context is not None
AttributeError: 'DisplayOpenGL' object has no attribute 'context'`
So i changed again back to Legacy Mode.

`$ sudo xinit /usr/bin/python3 /home/pi/pi3d_demos/PictureFrame :0`
I get `-bash: $: Command not found`
And with `sudo xinit /usr/bin/python3 /home/pi/pi3d_demos/PictureFrame :0`
`(EE)
Fatal server error:
(EE) Server is already active for display 0
        If this server is no longer running, remove /tmp/.X0-lock
        and start again.
(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE)`

I have to check the images sizes of the pictures. However there are ~5000 in the folders. Can you tell me for what i have to look? i can also post-process them.

paddywwoof, 2020-02-29:
Hi, wow that's lots of info - sorry you're having to mess around so much, and yes the pixels (width) probably are the critical thing. The Fake KMS is the new graphics driver and it really goes along with the RPi4 and requires the desktop (or xinit etc). On the zero it's probably best to use the legacy driver and not start the desktop. Probably some of the pictures have missing exif info and they're the ones causing the messages - they're just helpful debug messages rather than errors!

The PictureFrame app should already be doing post-processing and resizing the images to fit the screen before attempting to display them (see https://github.com/pi3d/pi3d_demos/blob/master/PictureFrame.py#L110). However there is a subtle difference between BLUR_EDGES True and False. If True then the image is scaled to your screen size and NOT scaled to one of the magic sizes (i.e. see https://github.com/tipam/pi3d/blob/master/pi3d/Texture.py#L25) that the GPUs on the RPi0,1,2,3 require.
  What happens if you set BLUR_EDGES False?

Thanks for helping sort out this problem.

PS if resizing the images did turn out to solve the problem you could do it with a very simple python program only a few lines long (which I would write for you) so it wouldn't be too much effort, even with 5000 images!

sl4m01, 2020-02-29:
With BLUR_EDGES = False it seems to work fine. However with the blured edges it obviously looked cooler ;-) but I think it's fine for now. Thank you a lot for your support!

paddywwoof, 2020-02-29:
I only asked you to set it to False to see if it was the resizing issue, which it certainly seems to be. Fixing it is simple a matter of changing that line, which I have already done on the develop branch. I will run tests over the next day or so then push it out as a new release. In a couple of days you should be able to
`pip3 install pi3d --upgrade`

sl4m01, 2020-03-01:
Super, thank you. Please close this issue or publish a short note here as soon as you are finished, then I will also notice it ;-)

sl4m01, 2020-03-01:
Like yesterday, also today the slideshow crashed  after running ~2h. I am only using it since yesterday. Didn't find an error log till now. Is there one?

paddywwoof, 2020-03-01:
That doesn't sound good. What form does the crash take? i.e. is it just python that stops (in which case I would expect a standard listing of what went wrong, so I assume that isn't what happens) or does the process stop responding and you have to kill it, or it carries on running but nothing appears on the screen, or the whole RPi freezes, or whatever...

Also are you starting python automatically from boot or manually in a terminal. Is the desktop running or are you booting to command line?

It's possible to write info to a log (which is what ought to  happen in the PictureFrame app rather than printing to screen) but it's quite hard to catch something that kills python without generating the normal termination dump. Some GPU related things do that.

Paddy

sl4m01, 2020-03-01:
Hi, sorry for the little info.
After some hours (not always the same) the slideshow just stops and the rasbian desktop appears. After a ssh reboot slideshow starts again.

I boot to desktop. Slideshow is set to autostart after reboot in crontab. Sh file to start is 
`#!/bin/bash
cd pi3d_demos
python3 /home/pi/pi3d_demos/PictureFrame.py`

Maybe it's possible to check every 5 min if the slideshow is still running, and if not do a reboot/restart of the script.

paddywwoof, 2020-03-01:
Hi, yes, I suppose it would be possible to monitor and restart it but it would be nice to figure out what was causing the problem. I'm not sure that it will work but if you start the program manually, maybe it will echo an error message to the terminal. i.e. ssh then
```
cd pi3d_demos
python3 PictureFrame.py
```
You might need to keep the ssh link open in order to not break the 'pipe'.

The other thing I was thinking of was to put a line in the program to write each image name as it's loaded, that way it might be possible to pin down if it's a specific image causing the problem. For that to work when the program loads with crontab it might be necessary to write the info into a file. I will make a test program.

sl4m01, 2020-03-03:
Hi, i just started the slideshow manually by the ssh command above today.
After the "usual" warning regarding exif und "glGetError 0x500" i receive after ~2h the line "Getoetet" which is German and means "Killed" in English. (i am using German in rasbian). After this the slideshow is stoped and is jumping back to the rasbian desktop. I can manually start it by ssh again. Dont know if this is helping.

sl4m01, 2020-03-23:
Hi paddywwoof, any news regarding this topic.
The picture frame is runing but of course it would be nicer with blured edges and without auto-restart after crashes. If you have anything new or need additional info please just let me know.
BR

paddywwoof, 2020-03-23:
Hi, Sorry for leaving this un-closed. The version of github.com/pi3d/pi3d_demos/PictureFrame.py from commit dcd67a8 (23 Feb) should work with blurred edges - let me know if it doesn't. The getoetet message sounds like a running out of memory error. What happens if you get just the largest images and copy them to their own temp folder and point PictureFrame PIC_DIR to that? (It's the WxH size that matters rather than the compressed jpeg file size but probably easier to sort by file size which will approximate)

closed 2020-08-28 by paddywwoof