pi3d/pi3d_demos #18: Specify Image Area?

jasoncross, 2020-06-28:
I’m running this on an ultra wide(aka bar) lcd for panoramic photos. It’s essentially a 1920x1080 display that has been cut down to 1920x560 with only the top part of the screen usable yet shows as a 1080p display to the OS. Is there a way to specify the image area on this so I can constrain images to the top area?

paddywwoof, 2020-06-28:
Hi @jasoncross, You can specify the display size when you create the Display object (using Display.create()) as well as the position of the top left corner, i.e. Display.create(x=0, y=0, w=1920, h=560) But there might be distortion if the display is telling the OS that it is a different size.

Are you using the PictureFrame2020 script? What computer/OS are you running on and, if raspberry pi linux, are you using the x server and which graphics driver? What happens at the moment? Does it run ok but the bottom half of the images is off the screen?

Sorry for all the questions - always curious to know what the cause and solution is to any pi3d problem!

Paddy

jasoncross, 2020-06-28:
I'm using the PictureFrame 2020 script.  Pi Zero W on linux.  I followed the instructions on https://www.thedigitalpictureframe.com/how-to-set-up-your-raspberry-pi-for-your-digital-picture-frame/

Right now, it looks like it's trying to fill the entire "screen" meaning I'm only seeing the top half of the images.

jasoncross, 2020-06-28:
I updated PictureFrame2020.py line 273 to include the width and height and that appears to work!  Would be nice if these were exposed in the config though.

idcline, 2020-12-03:
Paddy, great information. This is a must for installing a monitor in a frame.  Would you consider moving all four parameters into the config file for future releases?

paddywwoof, 2020-12-03:
@idcline, I will have another look at this. There are lots of changes in the the process of being rolled out (toggling name, date, location etc) so might as well add this if it makes a difference. Will probably need a bit more text in the help to explain why this might be needed.

idcline, 2020-12-03:
I just sent my "how I built my frame" story to Wolfgang today and noted this.  When ordering a frame and custom mat, frame shops always shrink the mat size a bit to allow for mounting uncertainty.  In addition to loosing a loosing a few percent of the image, I also found that mounting a monitor precisely in a frame is a challenge.  I used DISPLAY = pi3d.Display.create(**x=10, y=10, w=1892, h=1052**, frames_per_second=config.FPS, display_config=pi3d.DISPLAY_CONFIG_HIDE_CURSOR, background=config.BACKGROUND) to shrink the 1920x1080 HDMI display and to shift it a little to align with my custom mat.  Without adjustment, many of my images had some important parts clipped.

I had to create a simple test target to display while experimenting with the above values.

paddywwoof, 2020-12-11:
This is now in config

closed 2020-12-11 by paddywwoof