pi3d/pi3d_demos #19: Request: More Transitions

qwksilver, 2020-07-25:
it would be nice to operate with some transition effects, and a random option.
my ideas would be 

1. to paint over 4-12 "polaroid" photos at slightly random angles all from the library, then cross fade on last+1 photo
2.  wide line fade horizontal/vertical/diagonal left and right
3.  narrow line fade
4.  startrek transporter fade (big white star that slowly eats the existing image and lands a new one)
5.  swisschease fade, with new image under old that get's eaten away
6.  page turn *any corner or side"
7.  so many more, incorporate some modern ideas from video editors here

paddywwoof, 2020-07-25:
Good idea. Some of these suggestions probably require a different approach to the transition mechanism. At the moment two Textures are passed to the shader which controls how the two are combined. 2,3,4 and 5 lend themselves to this, you "just" need to devise an algorithm that parameterises the foreground/background proportion as a function of x, y, t

idea #6 probably needs something more sophisticated as the foreground image needs to be distorted as the page corner or side is "lifted"

idea #1 sounds like you want to show more than two images at once and I think the GPU can only cope with 8, quite apart from the issue of running out of memory etc. To do this you probably need to either overwrite an image using PIL or switch glClear() off (probably just the depth but this would need a facility adding to the internals of pi3d).

Other transitions I've thought about are origami paper folding or water running down a window.

One thing I would say is that when Wolfgang first started working on the PictureFrame demo he was keen to use all kinds of interesting fade effect, however after using it for a while he became convinced that really the only one that should be used long term was the simple fade!

qwksilver, 2020-07-25:
#1 would look like this, with the first background image being left behind, and each image appearing to fly/float/flutter/fade in.  

https://www.google.com/search?q=polaroid+collage&sxsrf=ALeKk00U3d0bXTCWtZDrKQ3ndfmCDiUS6Q:1595701375270&source=lnms&tbm=isch&sa=X&ved=2ahUKEwiJyo_tgunqAhUEa80KHfLYBW4Q_AUoAXoECA4QAw&biw=1918&bih=2029#imgrc=ZPB2bSKegXWwdM

say 4 show up, then when it's time for photo #5 it fades in with a regular transition
then a random number of pages before the next set of polaroid, with config options for say 
use polaroid(true/false), #of images 4-(responsible limit), min number of skips, max number of skips, polaroid delay

then true, 6, 3, 7, 3 mean that polaroids will show up then 6 fall in(one every 3 seconds), then it will be between 3-7 "normal" transitions before the polaroid hits again.

qwksilver, 2020-07-25:
(2-5) would be nice to see soon.  the one way is decent, but in time not so artful, and i've built(66%) an art frame that will eventually host my feature photos.  more ideas like (4) but with say.  a basic version of 6 could be just a corner diagnal fade with a harder fade line.  i would to see some 3d stuff too, though like wrapping around a ball that bounces to bottom and then off screen, or a black hole that uses a random point on the screen to "suck" the photo through, honestly most of these things are in adobe premier, and other video editors.