pi3d/pi3d_demos #20: Request: Rotation

qwksilver, 2020-07-25:
in a desktop free configuration setting rotation and keeping the 3d graphics processor is hard, right now I pre rotate images, but it's sloppy and I have to do more preprocessing, adding a config item to set the display orientation and have the software rotate, or reset the screen would be very nice.  thanks.

paddywwoof, 2020-07-25:
Not really sure what you mean by this. Could you give me some more specifics - maybe an image or two that causes problems and an example of the config setting that would be needed to correct it. PictureFrame should look at the image exif data and rotate accordingly.

qwksilver, 2020-07-25:
no photos right away, but the physical monitor is 90 deg anti-clockwose, in portrait mode.  i don't use the desktop components on the pi, so......i have no way to tell the pi 4 that it's rotated right.  it probably processes the exif data fine, just not the direction i need.  right now i have to build pre rotated files.  here is the basic idea (from another person)  https://www.google.com/search?q=photo+frame+portrait+digital&tbm=isch&ved=2ahUKEwiQkfqOh-nqAhUJb60KHbfGCNQQ2-cCegQIABAA&oq=photo+frame+portrait+digital&gs_lcp=CgNpbWcQAzoCCAA6BAgAEB46BggAEAUQHjoGCAAQCBAeULAjWKksYKwuaABwAHgAgAG4AYgBlQeSAQM0LjSYAQCgAQGqAQtnd3Mtd2l6LWltZ8ABAQ&sclient=img&ei=93wcX9C5OonetQW3jaOgDQ&bih=2029&biw=1918#imgrc=lyzpJ89iTo7-5M

paddywwoof, 2020-07-25:
Ah, I see that makes total sense now. Have you tried setting the screen orientation in boot/config.txt like this https://www.raspberrypi-spy.co.uk/2017/11/how-to-rotate-the-raspberry-pi-display-output/#prettyPhoto (this works fine for my monitor/raspberry pi) that way it would be the right way round for everything, not just the images.

qwksilver, 2020-07-25:
that doesn't work on pi4, something to do with the gpu that takes over, and if i disable the gpu the rotation works, but pi3d pitcures doesn't

paddywwoof, 2020-07-25:
OK I will check that out. There must be a way of setting the orientation the display surface rather than rotating everything that needs to be displayed!

paddywwoof, 2020-09-11:
Bert has let me know that he got this working:

<<<
I tried to use ```xrandr -o left``` via ssh after pictureframe2020 was loaded but didn't work, afterwards I found another bash command ```DISPLAY=:0 xrandr --output HDMI-1 --rotate right```, that can rotate the screen, that did work, but the problem was that half of the screen was black, a big border. The only way i could make it work good, was when i stopped the pi3donpi4.service, then started it again and immediately afterward i gave the ```DISPLAY...``` command. Then pi3d used the full screen in left orientation.
I wrote another service and a small script to make it work directly.
Because it took me a while to make it work and search half of the internet for it, i thought i could share it with you.
I'm not an power linux user, maybe there's an more elegant solution, I don't know

--------vertical.service----------
```
[Unit]
Description=Screen-rotation
After=multi-user.target pi3donpi4.service

[Service]
Type=Idle
ExecStart=/usr/local/bin/vertical_view.sh

[Install]
WantedBy=multi-user.target
```
-----------vertical_view.sh---------
```
#!/bin/bash
sleep 0.5
DISPLAY=:0 xrandr --output HDMI-1 --rotate right
exit
```
Also I chmodded both the service and script with 644, normally i just use chmod +x for script files, but I decided to use 644 this time to be sure.
>>>
I've not tried this yet as I'm away from home but many thanks to Bert for this contribution.

closed 2020-09-11 by paddywwoof