pi3d/pi3d_demos #21: Bug/Clarification

qwksilver, 2020-07-25:
I thought I read that it will restart or reindex/rerandomize a directory if a file is missing.
I just changed a file and removed an old one and it froze completely (10 min or more)
my directory was only 10 files at 16 sec rotation, so....should complete the file list after < 160sec or 3 min ish and reindex and redisplay.  I thought, but it just stayed put on that one file.

the request here is this, if it goes to load a file, and finds it missing, could you have it jump straight to reindex like it does if the directory changes contents.

thanks

paddywwoof, 2020-07-25:
It sounds odd that it froze for such a long time when just one file had been deleted. The file loading happens in a try/except block so it should generate an error message along the lines of "Couldn't load file {} giving error: {}" and try the next file in the list. I will check this out here but if you could do some more experiments that might be helpful.

The default file checking happens every 60s at the moment (which seems rather often now I think about it) but it does leave an opportunity for the next picture to try to load before the directory is rechecked. At the moment there is no kind of locking so maybe a new list is being generated while the image loader is trying to read one and some kind of race happens.

I will look a bit more at this but it could well be a good system to trigger a reindex if a file is missing.

qwksilver, 2020-07-25:
i would love for it to check the whole director before each photo change starts, missing file, or new files, reindex.  this would let me use dropbox/rclone to hot drop from my camera durring a shoot.  i need to speed rclone up somehow, but besides that, it could let me build a nice second tether screen.

paddywwoof, 2020-08-28:
should now be fixed

closed 2020-08-28 by paddywwoof