pi3d/pi3d_demos #22: Jagged text when using PictureFrame2020.py

mgainesdev, 2020-07-28:
I'm using PictureFrame2020.py to show movie posters on a vertical LG TV. It works, except text shows up very jagged. If I open the same file in the Pi default picture viewer, or if I run Windows on the same TV, the images look fine. The text is also jagged in the VNC viewer when running PF2000.

These two pictures are from the same LG TV (1080x1920 vertical orientation) and the same Pi.

I've tried everything I could think of from the docs and nothing seems to be able to make the image's text smooth. Any ideas?

[image]
[image]

paddywwoof, 2020-07-28:
Hi, this sounds very similar to the problem posted on the Raspberry Pi forum by @starmike https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=280854&p=1702515&hilit=pi3d#p1702515 (not responded so I don't know if it helped or not). If you try the suggestions on that link it would be very helpful if you posted the results back here in case anyone else runs into the same issue.

Is the HDMI set up to be rotated 90 degrees or are the images already rotated? Also what dimensions are the images to start with? The fact that 1080 is one of the standard GPU texture widths makes the scaling issue seem less likely. I suppose, alternative to the suggestions mentioned on the link above, it could be worth using a higher quality PIL.Image.rescale() (PictureFrame2020 line 62 or 60 (depending on HDMI rotation)) The default is BICUBIC but you could try ``...), resample=Image.LANCZOS)`` Obviously the higher quality resampling during resize will only have any effect if you *don't* use the ``--auto_resize=false``

Let me know how you get on.

mgainesdev, 2020-07-28:
Hi, Paddy. That was my post on the pi forums. I didn't see your response and after a few days I thought it got buried, and I figured it was more of a pi3d issue so I posted here.

Also, apparently, I have an account on here with @starmike which I completely forgot I made since I use git mostly for work.

I tried everything you said and nothing worked. Let me answer your questions:

The HDMI is rotated 90 degrees so that the TV is viewed vertically.
The TV is HDTV, so the viewing resolution is 1080x1920.
Some images are 1080x1920, some are much larger. The Starfigher image is 2000x3000. The Star Wars poster is 1981x3000. I seem to have the most trouble with text with images that are larger than 1080x1920.

The image to the Star Wars poster is here if you want to try it yourself:
https://www.starwarsarchives.com/wp-content/uploads/2017/01/EP4-PS-001.jpg

Using samples=4, it made the fading transition much slower, but the text didn't improve.

Once I determined that the problem was with PF2000 and not the HDMI, I moved my question here since this seems to be the root of the issue.

paddywwoof, 2020-07-30:
@mgainesdev sorry not to get back to you Yesterday, also thanks for sending the jpg. At the moment all I can confirm is that I can  reproduce the problem and I've tried the various things I suggested plus more and can't really make it much better. I think it must be something to do with the resizing process so I'm trying some different things with that. I will let you know what I come up with.

mgainesdev, 2020-07-31:
Thanks, Paddy. No rush, I'm just curious what the fix is.

mgainesdev, 2020-08-27:
I fixed it. You have to add `,Image.ANTIALIAS` to the lines that have im.resize

paddywwoof, 2020-08-27:
@mgainesdev Well done. It was down on my list of jobs to figure out how the scaling in the default image viewer (feh?) differed from PIL resize; the other puzzle was why it looked fine on this ubuntu laptop. I will add a comment to the PictureFrame2020 code for anyone in the same postion. (I will also look at the PIL source to see how they do the antialiasing).

Thanks again

Paddy

paddywwoof, 2020-08-27:
@mgainesdev could you just clarify your solution explicitly. In the PIL docs they say that PIL.Image.ANTIALIAS is just an alias for PIL.Image.LANCZOS. Was it just that this needed to be done in the lines ```if w > max_dimension``` (which it obviously does. Doh!)

mgainesdev, 2020-08-28:
This is what I did:

`    if w > max_dimension:
        im = im.resize((max_dimension, int(h * max_dimension / w)),Image.ANTIALIAS)
    elif h > max_dimension:
        im = im.resize((int(w * max_dimension / h), max_dimension),Image.ANTIALIAS)`

I did read about the alias, so I suppose you could use LANCZOS. I only used ANTIALIAS because it was in the sample code I wrote, and it matched a Windows version I worked on also.

closed 2020-08-28 by paddywwoof