pi3d/pi3d_demos #23: Slow loading PictureFrame2020.py

djeantou, 2020-08-09:
It is really slow to start up and load my picture collection. I believe this is because it is trying to load exif data for each photo before starting. 

Is it possible to check the exif data for each image before loading it on screen, to save time? Or to cache exif data from a previous run?

paddywwoof, 2020-08-09:
Hmm, I'm sure it would be easy enough. The reason it reads it on start up is so the date filter can be used but I should think it's no problem to make that toggleable.

djeantou, 2020-08-10:
Thanks. 
As an update to this I have a folder of 7586 photos and it takes about 30mins to load PictureFrame2020.py. feh loads the first photo straight away (though this is way better, espically if the load times can be brought down).

paddywwoof, 2020-08-10:
30s would be annoying! I think it would be easy enough to delay the exif reading until the image needs loading as a config option or something, I will try and knock something up tonight so you can try it.

Paddy

paddywwoof, 2020-08-10:
@djeantou I've tried moving the exif reading into its own little function and made that conditional on a config option. It seems to work OK (nearly instantaneous on this laptop) but I would be grateful if you could give it a try. The issue with this method is that the date_from or date_to could be set to exclude all images and there would be no error message - it would just run through trying all the images again and again. I need to think of a way to check that - maybe the orientation could be initially set to -1 or something to act as a flag - then filled in with the exif once read. Here is the code I've change in PictureFrame2020.py

```python
#####################################################
# some functions to tidy subsequent code
#####################################################
def tex_load(fname, orientation, size=None):
  global date_from, date_to
  try:
    im = Image.open(fname)
    if config.DELAY_EXIF:
      (orientation, dt) = get_exif_info(fname, im)
      if date_from is not None:
        if dt < time.mktime(date_from + (0, 0, 0, 0, 0, 0)):
          return None # try next slide TODO potential loop forever if no slides within date
      if date_to is not None:
        if dt > time.mktime(date_to + (0, 0, 0, 0, 0, 0)):
          return None # TODO check for infinite loop

    (w, h) = im.size
    max_dimension = MAX_SIZE # TODO changing MAX_SIZE causes serious crash on linux laptop!
    if not config.AUTO_RESIZE: # turned off for 4K display - will cause issues on RPi before v4
        max_dimension = 3840 # TODO check if mipmapping should be turned off with this setting.
    if w > max_dimension:
        im = im.resize((max_dimension, int(h * max_dimension / w)), resample=Image.LANCZOS)
    elif h > max_dimension:
        im = im.resize((int(w * max_dimension / h), max_dimension), resample=Image.LANCZOS)
    if orientation == 2:
        im = im.transpose(Image.FLIP_LEFT_RIGHT)
    elif orientation == 3:
        im = im.transpose(Image.ROTATE_180) # rotations are clockwise
    elif orientation == 4:
        im = im.transpose(Image.FLIP_TOP_BOTTOM)
    elif orientation == 5:
        im = im.transpose(Image.FLIP_LEFT_RIGHT).transpose(Image.ROTATE_270)
    elif orientation == 6:
        im = im.transpose(Image.ROTATE_270)
    elif orientation == 7:
        im = im.transpose(Image.FLIP_LEFT_RIGHT).transpose(Image.ROTATE_90)
    elif orientation == 8:
        im = im.transpose(Image.ROTATE_90)
    if config.BLUR_EDGES and size is not None:
      wh_rat = (size[0] * im.size[1]) / (size[1] * im.size[0])
      if abs(wh_rat - 1.0) > 0.01: # make a blurred background
        (sc_b, sc_f) = (size[1] / im.size[1], size[0] / im.size[0])
        if wh_rat > 1.0:
          (sc_b, sc_f) = (sc_f, sc_b) # swap round
        (w, h) =  (round(size[0] / sc_b / config.BLUR_ZOOM), round(size[1] / sc_b / config.BLUR_ZOOM))
        (x, y) = (round(0.5 * (im.size[0] - w)), round(0.5 * (im.size[1] - h)))
        box = (x, y, x + w, y + h)
        blr_sz = (int(x * 512 / size[0]) for x in size)
        im_b = im.resize(size, resample=0, box=box).resize(blr_sz)
        im_b = im_b.filter(ImageFilter.GaussianBlur(config.BLUR_AMOUNT))
        im_b = im_b.resize(size, resample=Image.BICUBIC)
        im_b.putalpha(round(255 * config.EDGE_ALPHA))  # to apply the same EDGE_ALPHA as the no blur method.
        im = im.resize((int(x * sc_f) for x in im.size), resample=Image.BICUBIC)
        im_b.paste(im, box=(round(0.5 * (im_b.size[0] - im.size[0])),
                            round(0.5 * (im_b.size[1] - im.size[1]))))
        im = im_b # have to do this as paste applies in place
    tex = pi3d.Texture(im, blend=True, m_repeat=True, automatic_resize=config.AUTO_RESIZE,
                        free_after_load=True)
    #tex = pi3d.Texture(im, blend=True, m_repeat=True, automatic_resize=config.AUTO_RESIZE,
    #                    mipmap=config.AUTO_RESIZE, free_after_load=True) # poss try this if still some artifacts with full resolution
  except Exception as e:
    if config.VERBOSE:
        print('''Couldn't load file {} giving error: {}'''.format(fname, e))
    tex = None
  return tex

def tidy_name(path_name):
[... 56 lines cut ...]
    exif_data = im._getexif()
    #print('orientation is {}'.format(exif_data[EXIF_ORIENTATION]))
    dt = time.mktime(
        time.strptime(exif_data[EXIF_DATID], '%Y:%m:%d %H:%M:%S'))
    orientation = int(exif_data[EXIF_ORIENTATION])
  except Exception as e: # NB should really check error here but it's almost certainly due to lack of exif data
    if config.VERBOSE:
      print('trying to read exif', e)
    dt = os.path.getmtime(file_path_name) # so use file last modified date
    orientation = 1
  return (orientation, dt)
```
and in config.py

```python
...
parse.add_argument(      "--auto_resize",   default=True, type=str_to_bool, help="set this to false if you want to use 4K resolution on Raspberry Pi 4. You should ensure your images are the correct size for the display")
parse.add_argument(      "--delay_exif",    default=False, type=str_to_bool, help="set this to true if there are a large number of images and it takes too long on start")
args = parse.parse_args()

...
AUTO_RESIZE = args.auto_resize
DELAY_EXIF = args.delay_exif

```

paddywwoof, 2020-08-11:
Hi, I've put a new version including these changes plus a way to break out of the infinite loop. https://github.com/pi3d/pi3d_demos/blob/develop/PictureFrame2020.py

djeantou, 2020-08-11:
Thanks just tested your last commit and it works perfectly. 
I appreciate the speed of that fix =D

closed 2020-08-11 by djeantou

paddywwoof, 2020-08-11:
I will test it with the various MQTT commands and other config options but if it seems stable I might make --defer_exif=true the default option. Thanks for bringing the problem to my attention.