pi3d/pi3d_demos #24: HEIC images do not display

hubertron, 2020-08-13:
As modern iPhone's shoot in HEIC it would be ideal for PhotoGallery Python script to support display of HEIC images. A package exists here: https://github.com/david-poirier-csn/pyheif that could do it, but would take some refactoring of the existing demo here.

paddywwoof, 2020-08-19:
@hubertron sorry not to get back to you earlier. I will have a look through this and let you know what I think. There are a couple of mods to the PictureFrame2020 script so it makes sense to see how/if HEIC fits in.

paddywwoof, 2020-08-20:
@hubertron I've made a couple of minor mods to PictureFrame2020.py on the develop branch https://github.com/pi3d/pi3d_demos/commit/3d974d5c23dcfa8d4fa5e7c83587e054a8dbe41b and it works fine on this ubuntu laptop. However I have struggled to install pyheif on the Raspberry Pi. I installed libheif-dev and libde265-dev - libffi couldn't be found but libffi-dev was already installed... but gcc is unhappy. Have you managed to install it?

hubertron, 2020-09-22:
I have, but I have forgotten how I managed to do it, it was a pain. I am setting up a new Pi in a few weeks for a friend and will document it then with that branch code and get back to you.

hubertron, 2020-09-22:
Actually I do remember. I added Ubuntu to my list of sources temporarily via `deb http://archive.ubuntu.com/ubuntu vivid main restricted universe`

then did the install of pyheif: `pip install git+https://github.com/david-poirier-csn/pyheif.git` and was able to import that package. I'll have to mess with your branch another time, but I believe that got it installed and running in a hello world sort of environment.

sapnho, 2020-10-11:
Hi Steven, I tried your approach to installing pyheif but wasn't successful. The deb command gave me the error "-bash: deb: command not found". I am using a Pi4 with the latest Raspbian. Anything else I might need to do before? Thanks!

vinyli85, 2020-11-20:
Hi all,

I managed to install pyheif on my RP3b+.
`Successfully built pyheif
Installing collected packages: pycparser, cffi, pyheif
Successfully installed cffi-1.14.3 pycparser-2.20 pyheif-0.5.1`

But I cannot open/show any HEIC-Images in Pi3D. Following message is shown after `cd /home/pi/pi3d_demos && python3 PictureFrame2020.py`

`have you installed pyheif?
trying to read exif cannot identify image file '/home/pi/Pictures/IMG_2214.HEIC'
Couldn't load file /home/pi/Pictures/IMG_2214.HEIC giving error: 'NoneType' object has no attribute 'size'`

Any ideas?
Thanks

paddywwoof, 2020-11-23:
Hi thanks for this info, I will follow your instructions and test this on the RPi tomorrow. Out of interest did you test just the phyeif bit on its own?

```python
>>> import pyheif
>>> from PIL import Image
>>> im = pyheif.read("img_file_name")
>>> im.size
>>> im.save("img_file.jpg"
```

vinyli85, 2020-11-24:
Don't know really how to do that but will try to find out. :)

paddywwoof, 2020-11-24:
Hi, Sorry. Those `>>>` symbols are the prompt symbols you get when you start the python 'terminal' so it's a kind of shorthand for:
1. open a terminal from the desktop
2. in the terminal type:
```bash
python3
import pyheif
from PIL import Image
im = pyheif.read("/home/pi/Pictures/IMG_2214.HEIC")
im.size
im.save("img_file.jpg")
```
After typing `python3` you will get the >>> prompts so what you type is interpreted as python. To escape from there type Ctrl-z

paddywwoof, 2020-11-24:
I would love to know how you managed to install pyheif. I've now spent another chunk of time getting nowhere and will put it to one side again until I get some more info (or maybe bullseye is released in a year or two).
when I follow carsales instructions `sudo apt install libffi libheif-dev libde265-dev` I get `Unable to locate package libffi` though the other two are installed. When I run their `sudo pip3 install git+https://github.com/carsales/pyheif.git` I get compilation errors (symbols are not recognized so presumably a header isn't their). If I try adding the `deb http://archive.ubuntu.com/ubuntu vivid main restricted universe` line at the end of `/etc/apt/sources.list` (presumably???) then apt complains when I `sudo apt update`
```
Err:4 http://archive.ubuntu.com/ubuntu vivid Release
  404  Not Found [IP: 91.189.88.142 80]
Reading package lists... Done
E: The repository 'http://archive.ubuntu.com/ubuntu vivid Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
```
and pip3 install generates the same compile errors.

vinyli85, 2020-11-24:
I installed the 'libheif', 'libffi' from GitHub-Rep from source and additionally 'sudo apt-get install x265'.

Then it was possible to do 'pip install pyheif'

I also tried to your test with pyheif bit and got an error: 'ImportError: /home/pi/.local/lib/python3.7/site-packages/_libheif_cffi.abi3.so: undefined symbol: heif_image_handle_get_raw_color_profile'

So I am stuck here.  Maybe I give it a try later...

paddywwoof, 2020-11-24:
OK, following my own instructions (!) https://github.com/carsales/pyheif/issues/26#issuecomment-686496395 I can basically get to the same point you are. But that's where I was back in September and I tried quite a few options all resulting in the undefined symbol error. Until HEIC becomes more generally used maybe you should change the setting on your phone to save as jpeg (and you would have to convert the existing HEIC to jpeg, but that's just a couple of lines in addition to the code I posted above)

vinyli85, 2020-11-24:
Checked version of `libheif` installed and it is only `1.3.2-2` where GitHub says `1.9.1` is the latest. Don't know how to update. Maybe this is the reason for that error.

Also found this https://github.com/strukturag/libheif/issues/381

paddywwoof, 2020-11-24:
I downloaded and compiled the latest stable version. I got it by ```wget 'https://github.com/strukturag/libheif/archive/v1.9.1.zip'
``` And ```pkg-config libheif --modversion``` gives 1.9.1 where did you get yours from?

I don't think the issue you link to is the same as this. However I did think it could be worth using a command line convert for each image to put it into /dev/shm/ 'on the fly'. But when I try
```
heif-convert -q 90 ~Pictures/heif/sample1.heic /dev/shm/sample1.png
```
I get a similar missing symbol error (I didn't include jpg support when I compiled libheif)
```
heif-convert: symbol lookup error: heif-convert: undefined symbol: heif_check_filetype
```
maybe it's worth asking a question on the strukturag issues..

vinyli85, 2020-11-24:
> I downloaded and compiled the latest stable version. I got it by `wget 'https://github.com/strukturag/libheif/archive/v1.9.1.zip' ` And `pkg-config libheif --modversion` gives 1.9.1 where did you get yours from?

Got it also from there but I messed up my whole system now. Will give it a try tomorrow or so..

paddywwoof, 2020-11-24:
OK I can sympathize - linux dependency hell is, well, hell.
I posted an issue on https://github.com/strukturag/libheif/issues/387 see if anyone there knows what might be the problem

vinyli85, 2020-11-24:
It's working!
Your script which converts `.heic` to `.jpg` is working and also pi3d (PictureFrame2020.py) is showing `.heic`-Images.
I made myself some notes, will test this tomorrow and let you know.

vinyli85, 2020-11-25:
So I think I found the problem with those symbol errors and others when trying to show `.heif`images.
I set up my RPi3b+ 3 times today and now everything is running including `pyheif` together with `pi3d`.

It looks like the "root problem" is `libde265-0`which comes pre-installed on the Raspbian Image. So here are the steps I did to get everything up and running:

Fresh install of RaspbianOS with Desktop based on `Release date: August 20th 2020 Kernel version: 5.4`
After first boot I did `sudo apt-get update && sudo apt-get upgrade` and then following steps:

1. `sudo apt-get remove libde265-0`and also check if those are NOT installed `libheif1 libheif-dev libde265-dev`
2. `sudo apt install autotools-dev automake libtool texinfo x265`
3. `git clone https://github.com/strukturag/libde265.git`
	`./autogen.sh`
	`./configure --disable-dec265 --disable-sherlock265 --prefix /usr`
	`make`
	`sudo make install`
Make sure `libde265`is installed **before** `libheif`. The `--disable`options will not install those 2 examples.
4. `git clone https://github.com/strukturag/libheif.git`
        `./autogen.sh`
	`./configure --prefix /usr`
	`make`
	`sudo make install`
5. `git clone https://github.com/libffi/libffi.git`
	`./autogen.sh`
	`./configure --prefix /usr`
	`make`
	`sudo make install`
6. `pip3 install git+https://github.com/carsales/pyheif.git`

Then you can test it with this from @paddywwoof and there should be no error and the converted picture will be created.

> python3
>import pyheif
>from PIL import Image
>im = pyheif.read("/path/to/image.heic")
>im.size
>im = Image.frombytes(im.mode, im.size, im.data, "raw", im.mode, im.stride)
>im.save("/path/to/img_file.jpg")

Maybe somebody can test and confirm this. I don't think you need to set up you Pi from scratch, just check for step 1.

paddywwoof, 2020-11-26:
@vinyli85, thank you so, so much for this. These library objects are not trivial to compile, even on the RPi4, so hacking through it so many times must have been really grueling!

TLDR; **it works**

I basically followed your instructions but starting from an RPi4 with libheif already cloned and compiled and pyheif pip3 installed without errors - but not working.

1. apt remove as per your 1.
2. apt install as per your 2. (I think they were all already there) 
3. compile and install libde265 as per your 2.
4. I had already git cloned, compiled and installed this so I went into the libheif-1.9.1 directory and added an initial
```make clean```
then followed your steps.
5. because I had pip3 installed pyheif previously without errors I thought that libffi might not be required so I tried pyheif without and it worked (I also didn't re pip3 install pyheif). I don't know if I had somehow previously installed libffi but I don't think so.

Thanks again, I will write some kind of FAQ for the PictureFrame app and put a link to this (as well as all the other tips and cool improvements people keep adding)

Paddy

closed 2020-11-26 by paddywwoof

sapnho, 2020-11-26:
I followed your instructions until I got in 3.

> pi@raspberrypi:~ $ ./autogen.sh
> -bash: ./autogen.sh: No such file or directory

vinyli85, 2020-11-26:
The `git clone` creates a new folder. You have to `cd` into that folder and then do these comands.

paddywwoof, 2020-11-26:
Yes I nearly commented on that but the name of the folder normally includes the version num so might be more confusing to have explicit instruction with the wrong folder name!

sapnho, 2020-11-27:
Great, this works, thanks! I combined your commands into one, so instead of the many separate ones, you can just use this one.. :-)

`sudo apt-get remove libde265-0 -y && sudo apt-get remove libheif1 -y && sudo apt-get remove libheif-dev -y && sudo apt-get remove libde265-dev -y && sudo apt install autotools-dev automake libtool texinfo x265 -y && git clone https://github.com/strukturag/libde265.git && cd libde265 && ./autogen.sh && ./configure --disable-dec265 --disable-sherlock265 --prefix /usr && make && sudo make install && cd .. && git clone https://github.com/strukturag/libheif.git && cd libheif && ./autogen.sh && ./configure --prefix /usr && make && sudo make install && cd .. && git clone https://github.com/libffi/libffi.git && cd libffi && ./autogen.sh && ./configure --prefix /usr && make && sudo make install && cd .. && pip3 install git+https://github.com/carsales/pyheif.git`

luksan, 2021-06-30:
I built libheif without a custom libde265 and libffi as follows. It doesn't take as much time, especially on the Pi3A+ I use.

```
sudo apt-get install libffi-dev libde265-dev
wget https://github.com/strukturag/libheif/releases/download/v1.12.0/libheif-1.12.0.tar.gz
tar xzf  libheif-1.12.0.tar.gz
cd libheif-1.12.0 
./configure --disable-static --disable-go --disable-examples --disable-tests --disable-visibility --disable-rav1e --disable-x265 --disable-aom --disable-gdk-pixbuf
make 
sudo make install  # Installs into /usr/local
```

Also, I use pipx to install picframe in a virtual environment
```
sudo apt-get install pipx
pipx install --system-site-packages picframe
. .profile  # to update PATH
pipx inject picframe pyheif  # builds and installs pyheif in the picframe venv
```

paddywwoof, 2021-06-30:
@luksan Thanks for this info, that looks miles simpler. Maybe it works because you disable something in ```configure``` that was causing the problem but not actually needed. Or maybe the fixes committed to struktuarag/libheif over last 6 months (lots) have done the trick.

A virtual environment would have made a lot of aspects of testing and running python *much* easier. However I wanted to test pi3d on the most "vanilla" setup most RPi users would have i.e. with the minimal ```apt-get pi3d```