pi3d/pi3d_demos #25: Network buffer time?

iJeffG, 2020-08-26:
I have 750Gb of pictures on an NAS.  I have one parent folder, and then 30 “year” folders beneath it.  In each “year” folder, there are sub folders by event, and those are where photos reside.  I am creating a digital frame with RPi4 8Gb.

I mounted my NAS “parent folder” to a folder under home/pi/FamilyPhotos.   I use this folder as the default in PictureFrame2020Config.py.   When I start P3d, it seems OK, but after 10 minutes, no images begin to rotate.  

When I change the default path to one of the “years”, it takes a couple minutes, but then that folders images begin to be cycled through.

Any way to change/configure so this can handle larger folder structures?  Thoughts on what I could be doing wrong?
NFS mounted, Gb WiFi router, and both RPi4 and NAS are on the same network inside the house

paddywwoof, 2020-08-26:
@iJeffG this is something I've tried to fix recently. https://github.com/pi3d/pi3d_demos/issues/23 but at the moment it's only in the develop branch. I will try running it on different Raspberry Pis then push the changes to master. (I held off because I'm also trying to get heic images working too)

closed 2020-08-28 by paddywwoof

iJeffG, 2020-09-02:
Thanks, Paddywwoof.  Any recommendations for how I could alleviate this? If I put a few pictures in the root of this folder structure, would it pop a picture up while getting initialized?   Not asking you to fix it, but any thoughts on how I could improve my implementation would be appreciated

paddywwoof, 2020-09-02:
iJeffG, this should be fixed now in the master branch https://github.com/pi3d/pi3d_demos/blob/master/PictureFrame2020.py. You will need the new version of config https://github.com/pi3d/pi3d_demos/blob/master/PictureFrame2020config.py as well as that includes the --delay_exif option (which now defaults to True). If you've modified either of those files you need to keep a copy then transfer any mods over manually.

Let me know how you get on.

iJeffG, 2020-09-02:
Awesome.   I will let you know