pi3d/pi3d_demos #26: Idea: Random pic at start + random fade_time and delay_time

Brt120, 2020-09-12:
Hello,

I made a small change in the pictureframe2020.py script to select a random picture from start, I tried setting shuffle to zero but i always have the same pics at start.

Anyway this is what i changed, maybe it's an idea to implement in the config script, set random true or false, i didn't test it yet with subdirectories.

# images in iFiles list
nexttm = 0.0
iFiles, nFi = get_files(date_from, date_to)
**next_pic_num = random.randint(0, nFi)**
sfg = None # slide for background
sbg = None # slide for foreground

I also wrote a small bash script that i can start once pictureframe2020 is loaded, it chooses a random value between 2 given variables, so there's a bit more variation, I'm sending mqtt messages over network, if you want to run the script on the host you can change the ip to localhost. I'm thinking of trying to implement the blend type in this script aswell
At first i tried to import random in the pictureframe2020config file but it didn't work, I don't know much about python

This is the script, it needs some more work and finetuning,  like adding a few reads but you got the basic idea.

#!/usr/bin/env bash
for i in {1..100}
do
  a=30
  b=60
  c=1
  d=10
  time_delay=$((a+RANDOM%(b-a)))
  fade_time=$((c+RANDOM%(d-c)))
  sleep_time=$((fade_time+time_delay))
  mosquitto_pub -h 192.168.1.21 -m "$time_delay" -t frame/time_delay
  echo $time_delay
  echo $fade_time
  echo $sleep_time
  sleep 2
  mosquitto_pub -h 192.168.1.21 -m "$fade_time" -t frame/fade_time
  sleep $sleep_time
done
exit

Kind regards
Brt

Brt120, 2020-09-12:
I have made some other changes that might be interesting
Added a next, and random-next, both with mqtt

elif message.topic == "frame/back":
        next_pic_num -= 2
        if next_pic_num < -1:
          next_pic_num = -1
        nexttm = time.time() - 86400.0
      **elif message.topic == "frame/next":
        next_pic_num = pic_num + 1
        nexttm = time.time() - 86400.0
      elif message.topic == "frame/random":
        next_pic_num = random.randint(0, len(iFiles))
        nexttm = time.time() - 86400.0**
      elif message.topic == "frame/subdirectory":
        subdirectory = msg
        reselect = True

and

client.subscribe("frame/quit", qos=0)
    client.subscribe("frame/paused", qos=0)
    client.subscribe("frame/back", qos=0)
    **client.subscribe("frame/next", qos=0)
    client.subscribe("frame/random", qos=0)**
    client.subscribe("frame/subdirectory", qos=0)

Kind regards
Bert

paddywwoof, 2020-09-13:
Bert, thanks for these. It does seem very odd that the same picture appears first. Are the subsequent pictures also in the same order? Can you tell if the shuffle hasn't happened at all (in which case the images will appear in name order) or that it does happen but the random.seed() is always the same.

The import of random should reseed it using the system clock but if python and the module is loaded before the time is reset (over the network presumably) and the time to loadhappens to be exactly the same each start then

closed 2020-09-13 by paddywwoof

Brt120, 2020-09-15:
I'll do some more testing in the weekend probably, I keep you posted, it's kinda busy at work for the moment.

Greetings