pi3d/pi3d_demos #27: Support for skipping subdirectory and images smaller than a certain size

CyclistHomeCoder, 2020-09-18:
Hi,

Ia really like the PictureFrame app you have created, thanks!

I have made a modification so I am able to specify a minimum file size and also to be able to skip directories with a certain name. These two options are useful for me who runs it directly on my NAS (that generates a lot of thumbnails).

Would you be interesting in merging my changes or should I maybe fork?

BR

paddywwoof, 2020-09-18:
Hi Cyclist, It would be great to share your info but I'm not sure the best way; the app is already bursting at the seams with all the improvements added over the years. Maybe the tidiest way is for you to have your version on github (gitlab etc) and I could put a comment with a link in the PictureFrame file (there are a couple of other major changes that people have done that I ought to do the same with).

CyclistHomeCoder, 2020-09-19:
No problem, I created a lightweight fork of the just the picture frame here https://github.com/CyclistHomeCoder/PIcFrame.

paddywwoof, 2020-12-11:
Hi, at some point I might try a more comprehensive re-structure of the picture frame code (splitting out the display and control along the lines of https://github.com/helgeerbe/picture_frame probably) and I will bear in mind a mechanism for including ideas such as yours in a reasonably logical way.

In the mean time, hopefully, anyone wanting to do what you did can follow your link. Thanks again.

closed 2020-12-11 by paddywwoof