pi3d/pi3d_demos #32: Linux newbie: help required

Duzeper, 2020-11-19:
I am new to Linux so need help.
I don't want to keep my photos for PictureFrame2020 on the same sd card as the OS so I have put them on an external drive.
My question is - How do I map a Windows drive to this card (/media/pi/32 Gb Volume) so that, in future, I can copy my photos from my Windows PC to the Pi?
Also, although I have successfully mapped a drive to the /home/pi folder and I can read  the contents, I cannot write to it even though I have set permissions to 777

paddywwoof, 2020-11-19:
Hi I'm the other way round: it's so long since I used windows I can't remember much and the details are probably out of date.
However I have had similar issues with write permission on external drives on linux and the solution I found was to mount the drive with specific usergroup permissions. The drive has to be formatted fat but I think that's normal for windows compatible drives. When I plug a drive into the USB of the pi and the X desktop is running it automatically gets mounted so the first thing I have to do is unmount it, so even before that I have to find what it's called in /dev/... And in order to map it I have to have previously created a directory on my pi. So this is the order I do it:
1. one off job create a drive to map to
```bash
pi@raspberrypi:~ $ mkdir usbdrive
```
2. job needed each time the drive is inserted. First find the reference, then unmount it, then remount it with required permissions
```bash
pi@raspberrypi:~ $ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    1  3.7G  0 disk 
└─sda1        8:1    1  3.7G  0 part 
mmcblk0     179:0    0 29.8G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 29.6G  0 part /

pi@raspberrypi:~ $ sudo umount /dev/sda1
umount: /dev/sda1: not mounted.

pi@raspberrypi:~ $ sudo mount -t vfat -o rw,exec,uid=1000,gid=1000,umask=022 /dev/sda1 usbdrive
```
Programs you start can then write and execute from the drive as well as read. The drive shows as the directory /home/pi/usbdrive (or whatever you called it)
I don't know of windows GUI programs (I'm sure there will be some) but I think the standard command line scp will work from windows terminal/command line
```bash
scp C:/Users/Me/Pictures/NewPictures2020_11/* pi@192.168.0.22:usbdrive/pictures/2020
```
(where the destination directory will have to exist beforehand: `ssh pi@192.168.0.22` ... `mkdir usbdrive/pictures/2020` ...)

Duzeper, 2020-11-20:
Hi Paddy.
I have done all that and the drive successfully mount to the path you mentioned, thanks for that. I still have some issues however.

1. This also needs to be repeated after a restart. Is there any way to automount the drive at boot?

2. more importantly, I still do not have permission to copy files from Windows to /home/pi/usbdrive
I have mapped a Windows drive to  \\raspberrypi\pi\usbdrive\pictures and although I can read and open files, I cannot write to the folder. I get the message 'You need permission to perform this action'
I have also tried  your suggestion of scp from a Windows command line,
'scp D:\RaspberryPi\test.txt pi@192.168.0.191:usbdrive/pictures' After entering the raspberrypi password I get 'Permission denied'

Also, why, in the Pi file manager, does the drive appear as '32 GB Volume', not usbdrive?

Now my brain is starting to hurt. I have just successfully FTP'ed a file to /home/pi/usbdrive/Pictures with no issues with permissions.
Clearly I could use this method to do what I need, but I am confused as to why I have permission to FTP a file but cannot copy a file using any other method.

paddywwoof, 2020-11-21:
@Duzeper It's hard to figure out the quirks and intricacies of the different programs! On my linux laptop I can certainly scp files to my raspberry pi - maybe there's a security aspect that means it's not safe from a windows computer. (I will boot up my old windows machine and have a go later.)

The fact that ftp works ok is probably a random feature of some ancient component of the software rather than anything anybody planned. See https://m.xkcd.com/2347/

I did a quick search for `raspberrypi mount on boot` and there were some suggestions that you could edit `/etc/fstab` and put the mount intructions in there.

Duzeper, 2020-11-21:
Yes, I have found a method for editing /etc/fstab and it does appear to work.
Many thanks for your help. I'm not sure I would have worked out the syntax by myself. No, I am sure. I would not!
I'll carry on researching the permissions issue from a Windows mapped drive, but at least the ftp solution allows me to do what I need.
PS Any progress with my MQTT query?

closed 2020-11-25 by paddywwoof