pi3d/pi3d_demos #33: photoframe.py

photo-album, 2020-11-24:
Photoframe displaying (large) photo collection in 'shuffle of' mode always starts with the same photo so the playlist never gets through to the far end. 
I'd like to shuffle the _start_ image then from that image onwards be recursive through the folders and files. That would get all the images seen over a few days.  Imagine a holiday trip - you'd want to see the pictures it in the time sequence but you dont want to see the same trip each start up. 

Apologies if I  missed the obvious in my trawl through the documentation.

paddywwoof, 2020-11-24:
Hi, something like this cropped up before. I think the random seed relies on using time and that's always the same value when starting at boot (though it seems unlikely that it would be exactly the same). I will try to think of a better way to re-seed it and how you can make it run in order but starting at a random point.

paddywwoof, 2020-11-25:
Hi, I've made a few changes on other things and pushed those changes up to the develop branch. However I'm not sure about the shuffle start option. I think it would need to be added as (yet) another command line config variable, as some people might want to start from zero if they were showing images from a specific subdirectory for instance.

As a test you could modify PictureFrame2020.py (around line 330 in the previous version, 360 in the new one)
```python
# images in iFiles list
nexttm = 0.0
iFiles, nFi = get_files(date_from, date_to)
#next_pic_num = 0                     ##<comment out
random.seed()                         ##<add
next_pic_num = random.randint(0, nFi) ##<add
sfg = None # slide for background
sbg = None # slide for foreground
```
The extra complication is when you change the start and end dates or subfolder using MQTT the selection is refreshed and next_pic_num set to zero. If that's an issue then you would need to change the code in a similar way around line 290 (313 new)

Let me know if this behaves as you would expect.

Paddy

photo-album, 2020-11-25:
Thanks Paddy
Give me a couple of days and I'll  report back.