pi3d/pi3d_demos #35: Request: Posibility to fit the entire image on the screen by changing it's proportions.

cupid666bg, 2020-12-12:
Hello,

I have recently configured a Pi3D picture frame with the help of your articles.
It looks great!

I wanted to ad however a feature that is currently missing by default but I am afraid my python coding skills are close to none :)

I have tried some research on the subject but it has not gotten me too far :(

So What I am trying to to is find a way to have Pi3D stretch an image horizontally until it firs the screen, even if it will change it's original proportions, but not cropping it.

So I have set the --fit option to "True" and now I have nice blurred edges, but I am trying to go beyond that.

Most of my photographs have been made with a DSLR with aspect ratio 3 by 4 which does not correspond to the widescreen of my TV.

I am looking through the /pi/pi3d_demos/PictureFrame2020config.py, but I seem to have trouble locating the exact part of the code that can do that for me.

To go even further I would like to try and figure some condition that would only stretch certain images which have propostions closer to the TV screen, so that when horizontally stretched they would not look too strange.

Can you please help me with some wisdom?

Thanks in advance!

Angel

paddywwoof, 2020-12-13:
Angel, I doubt I can provide any wisdom, per se, but I can point you at the bit of code that controls the image aspect ratio. I'm not sure stretching images is what most people want in a picture frame so I probably won't modify the code to provide this as a config option but you can achieve it by editing around line 484 (in the latest version)
```python
    slide.unif[51:53] = slide.unif[48:50] # transfer front width and height offsets
    wh_rat = (DISPLAY.width * sfg.iy) / (DISPLAY.height * sfg.ix)
    if wh_rat >= 0.75 and wh_rat <= 1.334: # change to suit <<<<<<<<<<<<<<<<<
      wh_rat = 1.0 #                                        <<<<<<<<<<<<<<<<<
    if (wh_rat > 1.0 and config.FIT) or (wh_rat <= 1.0 and not config.FIT):
      sz1, sz2, os1, os2 = 42, 43, 48, 49
    else:
      sz1, sz2, os1, os2 = 43, 42, 49, 48
      wh_rat = 1.0 / wh_rat
    slide.unif[sz1] = wh_rat
```
i.e. wh_rat is the ratio of the screen aspect ratio (=DISPLAY.width/DISPLAY.height) to image aspect ratio (=sfg.ix/sfg.iy) so setting it to 1.0 forces them to match. However the difference is quite marked between a screen 1920x1080 (1.777) and image 4x3 (1.333) so the distortion will be visible.

Let me know how you get on.

Paddy

cupid666bg, 2020-12-13:
Hello,

Many thanks! Now finally there is progress!!!
So I have modified the following code:
    wh_rat = (sfg.iy) / (sfg.ix)
    print(wh_rat)
    if wh_rat >= 0.75 and wh_rat <= 1.334: # change to suit <<<<<<<<<<<<<<<<<
      wh_rat = 1.0 #                                        <<<<<<<<<<<<<<<<<

Now the 3 by 4 pictures are finally getting stretched!! But so are all the others :(
I have checked and it seems that the wh_rat comes with the same value at every iteration even if the pictures are completely different.
And I dont know why :(
Thats what I have from the print:
0.5625
0.5625
0.5625
0.5625
0.5625
0.5625
0.5625
0.5625
0.5625
0.5625
0.5625
0.5625
0.5625

Can you please help me get past that?

paddywwoof, 2020-12-13:
Well you didn't really need to do anything but it's good to see you experimenting. In the picture frame code wh_rat is the ratio of the screen aspect ratio compared with the image aspect ratio. i.e. when they are the same (you have an image 16:9 showing on a TV 1920:1080) wh_rat will be 1.0. But in your version of the code you have overwritten the correct value of wh_rat with the **_inverse_** aspect ratio of your image (see below).

In your example wh_rat seems to be always 0.5625 so it won't pass the test in your if statement, so it won't get set to 1.0 But it will be a long way from the original value calculated above so the image will be distorted when displayed.

Try changing back to the original code but inserting the two lines as per my snippet (marked ```# <<<<<<<<```) if you want to see what's going on you could do something like
```python
    scr_ar = DISPLAY.width / DISPLAY.height
    img_ar = sfg.ix / sfg.iy
    wh_rat = scr_ar / img_ar
    print(scr_arr, img_ar, wh_rat)
    if wh_rat >= 0.75 and wh_rat <= 1.334: # change to suit <<<<<<<<<<<<<<<<<
      wh_rat = 1.0 #                                        <<<<<<<<<<<<<<<<<
```

cupid666bg, 2020-12-13:
Hello,

Thank you for your prompt answer!
I have modified the code as you instructed, but now I seem to be getting all the same values and none of the pictures get stretched:

1.7777777777777777 1.7777777777777777 1.0
1.7777777777777777 1.7777777777777777 1.0
1.7777777777777777 1.7777777777777777 1.0
1.7777777777777777 1.7777777777777777 1.0
1.7777777777777777 1.7777777777777777 1.0

Do you think it may have something to do with the way that sfg is initialized?
As I wrote my python skills are none, but can you please check that line:

sfg = tex_load(pic_num, iFiles, (**DISPLAY.width, DISPLAY.height**))

This is on line 437.

 and write me if that might be the reason that 
(sfg.iy) / (sfg.ix) always equals DISPLAY.width / DISPLAY.height:

paddywwoof, 2020-12-13:
Ah, If you look at the code for tex_load you will see that passing the display width and height is only used for making the blurred bars on either side (or above/below) when there is a difference between the image size and the screen size. In those circumstances the image is generated by combining the actual image with a blurred copy the same size as the display. So the result is always the same size as the display.

If you want your image to fill the display then you will need to turn off the config.BLUR_EDGES. However that then makes it more complicated in the instances where you don't want to stretch your image but do want to use the blurred edge effect. See what you get turning it off and I will think if there's a simple way to swap between.

Paddy

paddywwoof, 2020-12-13:
Angel, having thought about this a bit more I think the easiest hack to do what you want is in the tex_load() function - around line 109. You will have to set --blur_edges=true
```python
    if config.BLUR_EDGES and size is not None:
      wh_rat = (size[0] * im.size[1]) / (size[1] * im.size[0])
      if wh_rat >= 0.75 and wh_rat <= 1.334: # change to suit      <<<<<<<<<<<<<<<<<
        im = im.resize(size, resample=Image.BICUBIC) #             <<<<<<<<<<<<<<<<<
      elif abs(wh_rat - 1.0) > 0.01: # make a blurred background. Now elif <<<<<<<<<
        (sc_b, sc_f) = (size[1] / im.size[1], size[0] / im.size[0])
        if wh_rat > 1.0:
          (sc_b, sc_f) = (sc_f, sc_b) # swap round
        (w, h) =  (round(size[0] / sc_b / config.BLUR_ZOOM), round(size[1] / sc_b / config.BLUR_ZOOM))
```
leave the later code (ln484) as it was originally. If you get resizing artifacts you could change the resizing type from Image.BICUBIC to Image.LANCZOS which is more processing intensive and slower but, as it happens while no fading takes place shouldn't be noticeable.

closed 2020-12-13 by paddywwoof