pi3d/pi3d_demos #40: PictureFrame2020: Show text is slow

Duzeper, 2020-12-19:
PictureFrame2020 is all working great, but I just have one slight niggle with the display text options.
Although they all work correctly, they are very slow to display the text. When I start them using MQTT, the frame refreshes straight away but the text slooowly fades up, taking around 20-30 seconds to fully display. I really only ever want the text displayed briefly, so is there a line in the code that I can adjust to bring up the text straight away?

paddywwoof, 2020-12-19:
You can control the time that the text is displayed for in config. It will fade in and out for an eighth of the time and remain solid for three quarters. The fade time is capped at 2s but the alpha of the fading in text is also capped at the same level as the fading in image. Some of these changes are subsequent to the last major update https://github.com/pi3d/pi3d_demos/blob/master/PictureFrame2020.py#L530

Duzeper, 2020-12-21:
Many thanks for that Paddy. It has fixed the issue.

Note to self - 'Check for updates before you ask questions!'

Once again, thaks to you, and Wolfgang, for all your help for an'ignorant' Windows user. It is true to say that virtually all of my, still extremely limited, Linux knowledge (and 100% of my Python knowledge) has come from this project.

Happy Christmas to you both. Stay healthy.

Martin

paddywwoof, 2020-12-21:
No problem. These 'issue' discussions are quite an efficient way to create FAQ documentation!

Happy Christmas to you too.

Paddy

PS I will convert this issue (and a couple of others probably) into a discussion, which is a new feature on github and seems more appropriate for this.

closed 2020-12-21 by paddywwoof