pi3d/pi3d_demos #44: Can nor use MQTT frame/delete, on a mounted SAMBA share

cupid666bg, 2020-12-22:
Hello,

So under a folder in my /pi/Pictures I have mounted a SAMBA share where my pictures reside.
This solution along with Pi3D works like  charm but I was trying to take it a bit further and install Mosquito, so that I can manipulate the contents of the frame including deleting the pictures I don't like.
I have however stumbled across a problem that I will need your help solving.
I have tested the installation with 3 Mosquito commands that I wanted to use frame/paused, frame/back, frame/delete.
If I set the pictures to be read from a folder residing on the SD card all 3 commands work fine.
But if I try that on the SAMBA share, only frame/paused, frame/back work, as frame/delete seems not to do anything.
Moreover once I use the "frame/delete", then all the other Mosquito commands stop working as well and in order to fix that I need to restart Pi3D.
Here is the command I use to mount the SAMBA share:

sudo mount -t cifs "\\\\192.168.1.3\\dela_i_dokumenti\\DELA I DOKUMENTI\\frame" "/home/pi/Pictures/NAS4frame" -o rw,file_mode=0**777**,dir_mode=0**777**,iocharset=utf8,context=u:object_r:rootfs:s0,,username=admin,password=********

I am using 777 as permission so I expect it is not an access issue.
I have even tried moving/deleting data from the mounted share using commands and it works, so I am clueless.
Can you please help me with some guidance how I could approach debugging the issue?
I see no error messages produced by Pi3D.

Many thanks in advance!

Angel

paddywwoof, 2020-12-23:
Angel, The delete code just uses the python os.rename() function which is basically a wrapper round linux mv so it ought to work I would have thought. Try the following and see what happens:
```bash
import os
move_to_dir = os.path.expanduser("/home/pi/DeletedPictures") # check that directory was created OK by previous attempts
os.rename("/home/pi/Pictures/NAS4frame/testpic1.jpg", os.path.join(move_to_dir, "testpic1.jpg"))
os.system("mv /home/pi/Pictures/NAS4frame/testpic2.jpg {}".format(os.path.join(move_to_dir, "testpic2.jpg")))
```
you can try that in a python3 teminal by typing ```$ python3``` see what works or what error messages you get. If it doesn't work you could try ```sudo mv```

cupid666bg, 2020-12-23:
Hello,
Many thanks for your prompt reply.
I have tested the "sudo mv" prior to oppening this issue and it works fine, but the issue in the frame still remains.
I will try the python code and let you know.

Thanks!

paddywwoof, 2020-12-23:
Even though you set the permission in theory to allow the user pi to do the deletion I think the fact that you mount with sudo might complicate it.

cupid666bg, 2020-12-23:
Hello,
I could not do a mount without sudo-ing.The os did not let me.Could you suggest an alternative mount command that would not require root permissions?

paddywwoof, 2020-12-23:
No I'm pretty sure you can't, but the fact that you have to use sudo possibly has consequences - or might not! If the ```sudo mv``` worked on the command line then it could well work inside ```os.system()``` I had to do that with the directory creation because the python wrapped version couldn't do some things (sudo make a directory with ownership of pi). So maybe the deletion should do that too to cope with SMB or mounted drives.

cupid666bg, 2020-12-23:
Update: Your code pointed me in the right direction. Many thanks for that.Here is the result:
Python 3.7.3 (default, Jul 25 2020, 13:03:44)[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> move_to_dir = os.path.expanduser("/home/pi/DeletedPictures")                                                                                                                             >>> os.rename("/home/pi/Pictures/NAS4frame/testpic1.jpg", os.path.join(move_to_d                                                                                                             ir, "testpic1.jpg"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 18] Invalid cross-device link: '/home/pi/Pictures/NAS4frame/test                                                                                                             pic1.jpg' -> '/home/pi/DeletedPictures/testpic1.jpg'
>>> os.system("mv /home/pi/Pictures/NAS4frame/testpic2.jpg {}".format(os.path.jo                                                                                                             in(move_to_dir, "testpic2.jpg")))
0

So only testpic2.jpg was moved.
I have googled OSError: [Errno 18] Invalid cross-device link:and I found an explanation like: os.rename only works if source and destination are on the same file system. You should use shutil.move instead.

Can you please advise me how to proceed?
Thanks in advance!
Angel

paddywwoof, 2020-12-23:
Angel, if mv worked then it's worth trying a fall back to that. Could you try editing PictureFrame2020.py around line 330
```python
      elif message.topic == "{}delete".format(id):
        f_to_delete = iFiles[pic_num][0]
        f_name_to_delete = os.path.split(f_to_delete)[1]
        move_to_dir = os.path.expanduser("/home/pi/DeletedPictures")
        if not os.path.exists(move_to_dir):
          #os.makedirs(move_to_dir) # creates directory owned by root if picture frame run with sudo so...
          os.system("sudo -u pi mkdir {}".format(move_to_dir))
        #os.rename(f_to_delete, os.path.join(move_to_dir, f_name_to_delete))
        os.system("sudo mv {} {}".format(f_to_delete, os.path.join(move_to_dir, f_name_to_delete)))
        iFiles.pop(pic_num)
        nFi -= 1
        refresh = True
```
i.e. comment out the os.rename and add the line ```os.system("sudo mv...```

cupid666bg, 2020-12-23:
There is some progress, but I have another problem now :)I initialy tested with mounting a folder with small ammount of pictires and they get propperly moved.Then I mounted a much larger folder (like 4 + 2 TB larger :D ) and now the pictures do get removed, but not moved.They do not go to the Deletedpictures folder.In fact I have no idea where do they go :(

cupid666bg, 2020-12-23:
Correction: They dont get deleted nor moved. Just the playlist is updated so it will not display them if I ask to see the previous image.

    On Wednesday, December 23, 2020, 8:34:44 PM GMT+1, Angel Georgiew <cupid666bg@yahoo.com> wrote:  
 
  There is some progress, but I have another problem now :)I initialy tested with mounting a folder with small ammount of pictires and they get propperly moved.Then I mounted a much larger folder (like 4 + 2 TB larger :D ) and now the pictures do get removed, but not moved.They do not go to the Deletedpictures folder.In fact I have no idea where do they go :(

paddywwoof, 2020-12-23:
Hi, are the drives identical apart from size? ie is the formatting the same? I had to format my usb drive as FAT to get it to allow execution from it (along with other specific mount settings) Don't really know if that's relevant.

Also do you have --verbose=true so you get any error messages?

cupid666bg, 2020-12-23:
I will check and let you know.Thanks

cupid666bg, 2020-12-23:
Both of the Samba shares have Ext4 format.I have Enabled the verbose trace and I found the culprit:Dec 24 00:02:27 Pi sudo[1476]: mv: target '/home/pi/DeletedPictures/_MG_2562.JPG' is not a directoryso I have made corrections to the following code and now it works!!!!        #os.system("sudo mv {} {}".format(f_to_delete, os.path.join(move_to_dir, f_name_to_delete)))
        os.system("sudo mv {} {}".format("'" + f_to_delete + "'", move_to_dir))
I have used the quotes for the f_to_delete because some path names have spaces in them which make the strings between the spaces being treated as separate arguments.
Million thanks for the help!
I really enjoy your product!
   On Wednesday, December 23, 2020, 9:40:18 PM GMT+1, Angel Georgiew <cupid666bg@yahoo.com> wrote:  
 
 I will check and let you know.Thanks