pi3d/pi3d_demos #45: PictureFrame2020 ignores --time_delay value in config file

jeffalperin, 2021-01-01:
I have `--time_delay` set to 60.0 in  PictureFrame202config.py, but my images seem to be displayed for a much shorter time period, about 8 seconds or so.  However, if I change time_delay via `mosquitto_pub -h localhost -t frame/time_delay  -m "60.0"`, that seems to work.

In case it matters, I have `--fade_time=10.0` and `--show_text=""`, and I'm using a 4k display.

Also, though probably unrelated, the `--kenburns=True` setting seems to be ignored unless I use a non-4k display.  This seems to be true even if I force the 4k display into 1920x1080 resolution; it still won't run the kenburns effect.

paddywwoof, 2021-01-01:
Hi, that does sound odd (both things) I will see if I can reproduce it and figure out the cause. I don't have a 4k display so can't reproduce that aspect - did you change any settings to account for the 4k display, such as setting it to not resize images? (Not sure that would matter for kenburns)
Paddy

jeffalperin, 2021-01-01:
Hi, Paddy.  Thanks for the quick response!  I've had quite a few issues that seem possibly related to the 4k display.  (Another example is that the transition animation is "jerky" on the 4k display, but quite smooth on my old lower res display.)  I have played with the auto-resize setting, but it doesn't seem to make a difference for the time_delay setting issue.

paddywwoof, 2021-01-01:
I've just tried a couple of things but not managed to see the effect you report. Could you try hacking PictureFrame2020.py around line 440 to write info to a log file - I'm assuming you're running the picture frame from boot with xinit
```python
while DISPLAY.loop_running():
  tm = time.time()
  if (tm > nexttm and not paused) or (tm - nexttm) >= 86400.0: # this must run first iteration of loop
    with open("/home/pi/logfile.txt", "a") as f: #                                                    <<<<<<<<< these two lines
      f.write("{:.2f}, {:.2f}, {:.2f}, {:.2f}\n".format(tm, nexttm, config.TIME_DELAY, time_delay)) # <<<<<<<<<<<<
    if nFi > 0:
      nexttm = tm + time_delay
      sbg = sfg
      sfg = None
```
If you run for a while with the wrong change times then send the mqtt message to fix it then see what's been written into `/home/pi/logfile.txt` and let me know. I will try and get to the bottom of the 4k issues once this strange problem has been sorted.

jeffalperin, 2021-01-01:
I'm running picture frame from ssh command line with the following:  `export DISPLAY=:0 && python3 /home/pi/pi3d_demos/PictureFrame2020.py -p /home/pi/4kpix`.  Does that make a difference?

jeffalperin, 2021-01-01:
here's the logfile contents:
```
1609538648.95, 0.00, 60.00, 15.00
1609538663.97, 1609538663.95, 60.00, 15.00
1609538678.99, 1609538678.97, 60.00, 15.00
1609538694.01, 1609538693.99, 60.00, 15.00
1609538709.02, 1609538709.01, 60.00, 15.00
1609538724.04, 1609538724.02, 60.00, 60.00
1609538784.07, 1609538784.04, 60.00, 60.00
1609538844.12, 1609538844.07, 60.00, 60.00
1609538904.16, 1609538904.12, 60.00, 60.00
1609538964.21, 1609538964.16, 60.00, 60.00
```

paddywwoof, 2021-01-01:
Thanks for doing that. It is curious: If you look in PictureFrame2020.py you will see that time_delay is set to config.TIME_DELAY near the start of the file and there shouldn't be anything to set it to 15.0 apart from an MQTT command. Could you search through the file and just check that there isn't anything else in there that might be setting it to 15. If there isn't then I wonder if mosquitto broker is sending that message - you could try running it without MQTT and see if it still gets reset.

If you are running that way through ssh then presumably you have the desktop with X11 server already running. As you mentioned 4k I assumed you were using a RPi4 with the fake kms driver.

jeffalperin, 2021-01-01:
Aha!  Setting `use_mqtt` to `False `"fixes" the issue (though at the cost of disabling mqtt functionality).  And I checked, but found nothing that hard-coded the value of 15 into `time_delay`; in fact, 15 doesn't appear anywhere in the file.

I am using rPi 4 (8gb) with the fake kms driver.  I installed the mosquito broker locally per Wolfgang's instruction, so I'm puzzled why it would be sending a message to set time_delay to 15.

paddywwoof, 2021-01-01:
Well at least it's narrowed it down a little bit. If you added a debug line in the MQTT section (around line 320 in the develop branch version)
```python
      elif message.topic == "{}time_delay".format(id):
        print("topic={}, message={}".format(message.topic, float_msg)) # << can print as you are starting from cmd line
        if float_msg > 0.0:
          time_delay = float_msg
```
Obviously you would need to switch use_mqtt back to True for this to execute. If you see a message arriving to set the time_delay to 15.0 then we just need to figure out why it's arriving. What process are you using to send the mqtt messages, just typing them on the cmd line via another ssh terminal? Or have you set up something on a phone?

If the message source is hard to track down then you could try sending an instruction to yourself with something like (about line 280):
```python
      client.publish("{}paused".format(id), payload="off", qos=0) # un-pause the slideshow on start
      client.publish("{}time_delay".format(id), payload=config.TIME_DELAY, qos=0) # try sending an instruction over MQTT!!! <<<<<
      if config.VERBOSE:
        print("Connected to MQTT broker")
```

jeffalperin, 2021-01-01:
I downloaded MQTT Explorer to try to debug the broker, and upon connection to the broker, it shows the following:
[image]
I'm a newbie with rPi, mqtt, etc., so I'm still not sure where these values are coming from (and note that `fade-time` isn't even valid; should be `fade_time`, I believe.)

paddywwoof, 2021-01-01:
Very odd. I'm not really up to speed with MQTT either to be honest. I ought to put some time in and learn how it all works.

There does seem to be a thing called retained messages which can cause problems. Here https://community.openhab.org/t/clearing-mqtt-retained-messages/58221 are some suggestions on clearing them.

jeffalperin, 2021-01-02:
I've done some investigating, and you're right -- it's the retained messages that are causing the problem.  (I think I might have set the "retain" flag when I was playing around with some iOS mqtt clients some time ago.)  MQTT Explorer has an option to delete retained topics/messages, and when I do that, the config file settings are obeyed.

Looks like it's also possible for a **client** to remove a retained message:  _The standard MQTT mechanism to clean up retained messages is sending a retained message with an empty payload to a topic. This will remove the retained message._ ([https://www.hivemq.com/blog/delete-mqtt-retained-messages-subtopics-recursively/](https://www.hivemq.com/blog/delete-mqtt-retained-messages-subtopics-recursively/)).  I did a brief test, and this seems to work.  Would it be useful to have PictureFrame2020.py "clear" any retained message the first time through the loop?

paddywwoof, 2021-01-02:
Yes, that's effectively what the openhub page suggested. I wondered about clearing the retained messages - it would break anything somebody had set up externally to use retained messages so I decided it was better to leave it external to the picture frame. I'm not sure how retained messages could be used but somebody might devise a clever system. Anyway, glad you solved it.

On the issue where kenburns doesn't work on 4k monitor I'm wondering if it's due to the image size. You said in your initial post that the config value is 'ignored' which implies that the slideshow runs but one of the fit or blur options is used for scaling. Is that the case?

To try to pin down the problem could you put a small number of images into their own subfolder and run just with those, using a 4k monitor. Run with kenburns off and kenburns on, as well as auto_resize on and off (i.e. 4 runs through)
To start with just have small images, maybe one 960x640 and one 640x960 or dimensions that don't fit your screen (otherwise the kenburns will have no effect). Then try increasing the size in stages up to the size of your normal pictures. If the small size still has problems then this isn't the cause of the problem and we need to look elsewhere!

Paddy

jeffalperin, 2021-01-02:
`Yes, that's effectively what the openhub page suggested. I wondered about clearing the retained messages - it would break anything somebody had set up externally to use retained messages so I decided it was better to leave it external to the picture frame. I'm not sure how retained messages could be used but somebody might devise a clever system. Anyway, glad you solved it.`  Makes sense.

`On the issue where kenburns doesn't work on 4k monitor I'm wondering if it's due to the image size. You said in your initial post that the config value is 'ignored' which implies that the slideshow runs but one of the fit or blur options is used for scaling. Is that the case?`  Yes.  `--fit = True`; `blur_edges = True`

`To start with just have small images, maybe one 960x640 and one 640x960 **or dimensions that don't fit your screen** (otherwise the kenburns will have no effect).`  I've been using images that DO exactly fit the screen resolution; could that be the problem?

`PPS what GPU memory setting are you using. `  GPU is 256

paddywwoof, 2021-01-02:
Hi, I suppose it's not exactly a problem but the way kenburns (on PictureFrame2020) works is to make --fit False i.e. the image is scaled so it just fits with no space then it moves so that the overlapping part gradually moves into the frame. So if the image is the same aspect ratio as the frame you shouldn't see any movement. If your images are already the same aspect ratio as the frame then --fit should have no effect and you should never see the edges to blur.  Can you give more of a description of what happens when the kenburns doesn't work?  Or do you mean that the number of pixels equals the width or height of the image but not necessarily the aspect ratio?

256 sounds like more than enough memory.

Paddy

jeffalperin, 2021-01-02:
I was cropping images to the exact resolution of the display (i.e. 3840 x 2160), so there's no edge blur showing (as expected) and no kenburns motion; the images just cross-fade with no movement (but I guess that's as designed, given your algorithm).  

I have some directories with images that I didn't bother to crop to 3840 x 2160, and those will display with blurred edges, and I guess those should show kenburns motion.  (I believe they do, but I need to double-check.)

So, to summarize:  since my display is 16:9 aspect ratio, if I crop to that same aspect ration, I won't see the kb animation because there is no "overlapping" part to move into the frame.  Is that right?

--Jeff

BTW, I've got some additional questions about 4k performance, text display, etc.  Would you prefer that I open separate issues for those or post in the discussions section or what?  Thanks so much for developing such a fantastic application and for so actively (and patiently) supporting it.

paddywwoof, 2021-01-02:
OK let me know what you see with the different images - and especially if you can see a pattern to ones that are inconsistent or jerkier etc.

It might be better to open another issue as the 4k performance is quite separate from the MQTT issue.

The app is really the result of people asking questions or making suggestions - all I'm doing really is fielding the and occasionally tweaking the code... case in point the whole thing probably needs a review and major re-write. But actually some of the users of PictureFrame have already done quite a lot of the thinking for that already so it won't be too hard for me to do.

Paddy

closed 2021-01-02 by paddywwoof