pi3d/pi3d_demos #49: Request: gif support

milika, 2021-01-02:
As title says, animated gifs would be great.

Thanks for all the great work

paddywwoof, 2021-01-03:
@milika this has cropped up a couple of times now see https://github.com/pi3d/pi3d_demos/discussions/37
I will knock something up that will show gif animations at some stage over the next week or two. I don't know what kind of FPS is achievable - probably not very high!

milika, 2021-01-03:
great :)
well i'm testing on pi zero w, and transitions have great fps...
How about mp4, or some other moving images format?
Thanks!

paddywwoof, 2021-01-03:
The normal graphics FPS don't include the time to load the image. On the zero that will be significant, especially for large images.

There is a demo rendering video (VideoWalk ?) It uses a very small video, you could try substituting a larger one and see what FPS you get. I would expect around 1fps. Obviously the picture frame code would need to be modified substantially to combine video and gif.

PS actually there was a project using video playback with pi3d overlay (foosball @swehner) I might look that up again.