pi3d/pi3d_demos #56: Feature request, PictureFrame2020 MQTT with host name in the topic

fanlessfan, 2021-01-22:
Hello, I setup two PictureFrame2020 servers with MQTT control. since the MQTT topic only starts with frame/#, there is no way to control them differently. Is it possible to add hostname in the topic?

thx

paddywwoof, 2021-01-22:
Hi, I think that's already in the develop branch https://github.com/pi3d/pi3d_demos/blob/develop/PictureFrame2020config.py#L49 then https://github.com/pi3d/pi3d_demos/blob/develop/PictureFrame2020.py#L332

However I am in the process of trying to refactor the whole thing so that might be relevant. The refactoring will be based on https://github.com/helgeerbe/

Did you have to alter anything in PictureFrame2020 in order to get it running on your RPi zeros?

fanlessfan, 2021-01-22:
Great. Thank you @paddywwoof . I didn't change anything in PictureFrame2020. I optimized piCore extension to avoid all the unnecessary ones.

paddywwoof, 2021-01-22:
Excellent. If you have any references or links that explain what you did I could include that in the pi3d FAQ

fanlessfan, 2021-01-22:
I will prepare one and will post it here when it's ready.