pi3d/pi3d_demos #60: Cursor Control

3tcubed, 2021-02-10:
Is there a way to turn the cursor on and off programmatically in pi3d?

I've reviewed source, and can't find an obvious method, here's what I've tried/investigated

I know you can turn it off when creating the display with DISPLAY_CONFIG_HIDE_CURSOR
but I want to turn it on again if a user hits a key sequence.

I tried setting DISPLAY.opengl.display_config = pi3d.DISPLAY_CONFIG_DEFAULT no luck
and tried defining and starting pi3d.Mouse still can't turn cursor back on

This is the last set of changes I had wanted to make before adding "PictureCalendar" (PictureFrame with Calendar functionality - Discussed with Wolfgang).

if I have to destroy the DISPLAY definition and start over it's not worth it

paddywwoof, 2021-02-10:
Well the hiding is done https://github.com/tipam/pi3d/blob/03e4e11ebd11ee05d907ae66e640e649761c680b/pi3d/util/DisplayOpenGL.py#L280 so that could be put into its own little function - I will look later to see what's involved with reversing it.

But it does occur to me that the relevant code is specific to an X11 display - on RPi3 and earlier without the desktop then I don't think there is a cursor at all. But there is a very simple pi3d.Gui system that you can see in the pi3d_demos Gui.py Which uses a little sprite as a cursor and you can toggle visibility of that simply by not doing draw(). That route might be more flexible but it might need a lot of re-writing for your mouse clicking functionality. Are you using a pi3d.Mouse?

Paddy

paddywwoof, 2021-02-10:
P.S. Timothy, I'm in the process of hacking the PictureFrame2020 code quite seriously as you will see on [this Discussion thread](https://github.com/pi3d/pi3d_demos/discussions/55). I don't know if that's relevant to your code or if it's too far off to be worth thinking about. The general objective has been to make it more modifiable...

3tcubed, 2021-02-10:
I've added the mouse interface, but since the cursor was off it did not
matter (not seen), had not considered a simple tracked sprite; I've added
lots of code that uses sprites. Really only looking at this because to add
info to calendar I open editor and with cursor off finding editor window
is.... challanging

I've also made lots of changes to PictureFrame to make it easier to change,
all variables that users can easily modify I've put in .cfg file with
relevent documentation.    I'd be happy to get you my source, the .cfg
handling come from code I've used for years (also had .ini processing thay
I pulled is it was not being used), having the code be more similar would
be useful.... let me know,  and thanks for the prompt response.

3tcubed, 2021-02-10:
Rename .zzz to .zip

I tried to upload these to pi3d, but it said I did not have permission;
I'll upload if you don't want to once I have permission, otherwise you may
In any case you might be interested in the changes I made to
PictureFrame.py for .cfg support.

The MonthMMMYYYY.txt is a sample format file of how a month's items are
added to the calendar, it's created on the fly when you hit enter on the
Calendar (for the selected month), cursor right/left to change the month
displayed (may take a second to respond, sampling of keyboard not done
frequently enough).  In any case, I'll be interested to know what you think.

paddywwoof, 2021-02-10:
Tim I think uploading to GitHub threads is very limited, probably only image files for the free accounts. You could upload it to your repo, or if you're not ready for that, email to patrick at eldwick dot org dot uk. I would certainly love to see your ideas, though if our code has diverged too much it may be quite a job to reincorporate everything!