pi3d/pi3d_demos #62: Config for PictureFrame2020

RichN001, 2021-02-24:
Hi, I have not tried this fully and just started looking at this solution, but I was wondering if the config file could be reloaded while PictureFrame2020 is running to accept changes on the fly?

I don't really use home automation and would rather control items from http locally and look on controlling functions from API. I like that you put orientation detect from EXIF, but images that are edited in Photoshop or other forms of Photographs that do not have this info we would have to manually flip the image to get it to work right. Maybe a option in config to rotate the display window and ignore EXIF orientation is worth having. (Rotating Rp4 from boot does not rotate PictureFrame2020. (I will look more in to that)

I have a script I wrote and run in Flask to upload images then have PictureFrame2020 look in the directory, maybe ill see if I can make some radio buttons for orientation that will re-write EXIF as my solution. Another reason I asked about config, thinking I could re-write config from my Flask App and they would take.

RichN001, 2021-02-24:
oh i read somewhere about getting a database in for images. maybe config can be read from there. I might look in to doing this and pull settings every xx seconds.

paddywwoof, 2021-02-24:
Rich, the whole system is in flux at the moment. see https://github.com/paddywwoof/picture_frame/tree/develop which has tried to split out the different parts of the functionality. This has allowed different interface classes to call methods on the controller i.e. there is a keyboard, mqtt and http interface. At the moment the interface_http is v. simple just using the http.server in python core. But making your own using Flask would be easy enough (if you are already happy writing Fask applications)

There isn't a method in controller to rotate an image but the rotation is stored in a db so if it was added, and a mechanism to write it back to the db (again thought about but not implemented yet) then the rotation would be permanent (or until there was a change that made the db record get re-written)

If you get a chance look at the new system (and the long discussion about it https://github.com/pi3d/pi3d_demos/discussions/55) and let me know what you think.

Paddy

RichN001, 2021-02-24:
just tried to get it going real quick got pi@raspberrypi:~ $ picture_frame
INFO:picture_frame.py:starting ['/home/pi/.local/bin/picture_frame']
INFO:model.Model:Open config file: /home/pi/.local/picframe/config/configuration.yaml:
INFO:controller.Controller:creating an instance of Controller
INFO:controller.Controller:Connected with mqtt broker
* failed to add service - already in use?

Ill take am look in a bit.. Thanks...

paddywwoof, 2021-02-24:
Rich, it's really good of you to have a go with the new project. It's based on helgeerbe/picture_frame but the working version is either in my develop branch ro Helge's dev branch (mainly try to keep them synced). I don't know how you tried to start it but it looks like it's pip installed itself in ```/home/pi/.local/picframe``` which makes me think it  might be an older or other  branch. For testing I think the best way (on RPi3 or less anyway):
```bash
clone repo
cd to repo
git checkout develop or dev (depending which repo cloned)
copy ./config/configuration_example.yaml up to to directory
edit and rename it so paths all point to relevant subdirectories of repo or Pictures and wherever you want to have your db
$ python3 -m picframe.picture_frame configuration.yaml
```
Or. There is an option to run ```picture_frame -i``` to install it properly Helge put instructions somewhere on the very long thread I linked to before.

RichN001, 2021-02-24:
ok, ill mess with it. Im on RP4 here, I can try on a 3 later too if I can't get it going.

paddywwoof, 2021-02-24:
On RP4 it's simpler if you have the X desktop running. Otherwise you have to launch it using sudo xintit etc and running as a module ```python3 -m ..``` runs into problems. If it's installed with pip3 to ```.local/lib/python3.7/site-packages``` then it's easier in some ways but probably causes problems if running as root. I need to make some clean installs and go through all the problems that appear when you don't happen to have something conveniently installed previously!

I've run it manually on RP4 by copying picture_frame.py up a directory (to ~/picture_frame) then running
```bash
$ python3 picture_frame.py configuration.yaml
```

RichN001, 2021-02-25:
Did not get enough time yesterday to get working. Im going to set up a new pi for testing here in a bit.

RichN001, 2021-02-25:
still no go, when I get some real time to sit with it Ill try again. I thought I went through the post pretty good to figure out any issues but still no go.. I didn't document anything.. next time I try Ill try to note it out.

paddywwoof, 2021-02-25:
Rich, I just updated to the latest and ran into a few wrong path issues. I will clear everything out tomorrow morning then re-download and do the ```python3 -m pip install -e``` then document and fix all the problems and write them up.

RichN001, 2021-02-26:
right on... thanks much. I had some path issues and think I got them straight.. then I had a problem with nine block not loading but it was installed... think I resolved that then a problem parsing the config... just every angle something. Im not one to give up but I may wait for you to get through it. Probably easier for you as you know it, im just trying to figure out where things are as I run in to them.. eventually Ill get there.. haha...

paddywwoof, 2021-02-26:
OK so starting with a new SD card using RPi4 and desktop running.
```bash
$ git clone https://github.com/paddywwoof/picture_frame.git
$ cd picture_frame
$ checkout develop
$ python3 -m pip install -e . ## NB dot on end puts it in same folder cloned and cd to
$ cp picframe/config/configuration_example.yaml ./configuration.yaml
$ nano configuration.yaml # you need to fix paths and put in some IDs etc. See below
$ python3 -m picframe.picture_frame configuration.yaml
```
my config file now looks like
```yaml
viewer:
  blur_amount: 12 
  blur_zoom: 1.0
  blur_edges: False 
  edge_alpha: 0.5 
  fps: 20.0 
  background: [0.2, 0.2, 0.3, 1.0] 
  blend_type: "blend" 
  font_file: "~/picture_frame/picframe/data/fonts/NotoSans-Regular.ttf" # <<<<<<<<<<<<<<<<<<<<<<<

  shader: "~/picture_frame/picframe/data/shaders/blend_new" # <<<<<<<<<<<<<<<<<<<<<<<
  show_text_fm: '%b %d, %Y' 
  show_text_tm: 20.0 
  show_text_sz: 40
  show_text: "title caption name date folder location" 
  text_width: 70
  fit: False 
  auto_resize: True
  kenburns: False 
  display_x: 0 
  display_y: 0 
  display_w: null 
  display_h: null  
  use_glx: False  

  mat_images: True  

  mat_type: null 
  outer_mat_color: null  
  inner_mat_color: null 
  outer_mat_border: 75 
  inner_mat_border: 40 
  use_mat_texture: True
  mat_resource_folder: '~/picture_frame/picframe/data' # <<<<<<<<<<<<<<<<<<<

  codepoints: "1234567890AÄÀÆÅÃBCÇDÈÉÊEËFGHIÏÍJKLMNÑOÓÖÔŌØPQRSTUÚÙÜVWXYZaáàãæåäbcçdeéèêëfghiíïjklmnñoóôōøöpqrsßtuúüvwxyz., _-+*()&/`´'•" # limit to 121 ie 11x11 grid_size

model:
  pic_dir: "~/Pictures" 
  deleted_pictures: '~/DeletedPictures' 
  no_files_img: "~/picture_frame/picframe/data/no_pictures.jpg" # <<<<<<<<<<<<<<<
  subdirectory: ""  
  recent_n: 7 
  reshuffle_num: 1
  time_delay: 200.0 # move deleted pictures here
  fade_time: 10.0
  shuffle: True  
  image_attr: [  
    'PICFRAME GPS',
    'PICFRAME LOCATION',
    'EXIF FNumber',
    'EXIF ExposureTime',
    'EXIF ISOSpeedRatings',
    'EXIF FocalLength',
    'EXIF DateTimeOriginal',
    'Image Model',
    'IPTC Caption/Abstract',
    'IPTC Object Name',
    'IPTC Keywords']
  load_geoloc: True 
  locale: 'en_GB.utf8'                    # "locale -a" shows the installed locales which could used
  key_list: [
    ['tourism','amenity','isolated_dwelling'],
    ['suburb','village'],
    ['city','county'],
    ['region','state','province'],
    ['country']]
  geo_key: 'rich_this_needs_to@be_changed_yr_email' #<<<<<<<<<<<<<<<<<<<<<
  db_file: '~/picture_frame/pictureframe.db3' # <<<<<<<<<<<<<<<<<<<<<<<<<<
  portrait_pairs: True                   # <<<<<<<<<<<<<<<<<<<<<<<<
  log_level: 'WARNING' 
  use_kbd: False 

mqtt:
  use_mqtt: False
  server: "localhost"              # <<<<<<<<<<<<<<<<<<<<<<<
  port: 1883                              #<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
  login: "                           # <<<<<<<<<<<<<<<<<<<
  password: ""               # <<<<<<<<<<<<<<<<<<<<<<<<<<
  tls: ""             # <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
  device_id: 'picframe' 

http:
  use_http: True                         # <<<<<<<<<<<<<<<<<<<<<<<
  path: "~/picture_frame/picframe/html"  # <<<<<<<<<<<<<<<<<<<<<<<<<<<
  port: 9000
  use_ssl: False
  keyfile: "path/to/key.pem"
  certfile: "path/to/cert.pem"
```
It threw an error when I set mqtt True (though it worked fine elsewhere - I will track down cause later. You should be able to access using html on 192.168.0.543:9000

Let me know how you get on (I'm sure you will)

Paddy

RichN001, 2021-02-26:
that did it... exactly as above, nites for someone else tying.. add git to checkout develop and I had to add a " to close  login in your sample config.. SO now its running... woohoo... thanks Paddy. 

I get below then it opens. http works.. have not dug in yet. 

pi@raspberrypi:~/picture_frame $ python3 -m picframe.picture_frame configuration.yaml
INFO:picture_frame.py:starting ['/home/pi/picture_frame/picframe/picture_frame.py', 'configuration.yaml']
INFO:model.Model:Open config file: configuration.yaml:
ERROR:model.Model:error trying to set locale to en_GB.utf8
Segmentation fault

RichN001, 2021-02-26:
just spotted local in config.. fixed that for me.

helgeerbe, 2021-02-26:
Hi Richi,

just to get it right. You took Paddys config file? I checked the provided `configuration_example.ymal` and that look OK to me. Maybe we should enhance the reading of the config file and catch the exception. 

When fixing the locale setting, you set it to something (like 'en_US.utf8') you configured  with `raspi-config`?  So again, we should catch the segmentation fault.

RichN001, 2021-02-26:
Hey, yeah I just copied his example above. With copying his example I took his local  locale: 'en_GB.utf8' setting. I changed it in the config to en_US.utf8. I see now looking in the config it is set already to en_US.utf8 so it would have been ok.

RichN001, 2021-02-26:
I basically just copied his to make sure I didn't miss a dir change.

helgeerbe, 2021-02-26:
OK, Thanks for the feedback. I think we have to point out in the user documentation to set the locale correctly. Normally no one cares about that, except he gets strange date formats.

RichN001, 2021-02-26:
is there anyway to grab that from the system? or just easier to put in config.

i guess closer to a release with a install.sh you could actually build the default config and pull locals for that and even set paths.

just thinking down the road of a zero touch install. certainly easy enough for now just to make a few config changes from documentation.

helgeerbe, 2021-02-26:
Good Point. I have something like that in mind with the -i option. What's missing is exactly not just copying 'configuration_example.yaml' but guide you through the config process.

paddywwoof, 2021-02-26:
Rich, just been out of contact for a while. I didn't explain that the # <<<<<<< type comments were showing lines that I had changed. On the locale line I left the original instructions i.e. you should run at a command prompt
```bash
$ locale -a
```
and that will tell you what to set your locale to. It it's wrong then there is a Raspberry Pi config option that you can run to make it work in your language.

I don't know if the paths in configuration_example.yaml are correct if you do an install without the -e. Maybe the install option will fix this.

Also I tried running on RPi4 from the command line (without desktop) but ran into all the old problems of needing to use sudo and then nothing being installed - there should be info on that in the other thread but obviously it needs  to be exhaustively tested and documented!!

RichN001, 2021-02-26:
As you instructed worked fine with being in desktop. It just stopped on those 2 things which where easy enough to sort.. Appreciate you helping me get it going.

paddywwoof, 2021-02-26:
It's good of you to have a play around with it. There are almost certainly bugs and the more people that look, the more likely we are to spot them.

RichN001, 2021-02-26:
yeah no problem, ill input as much as I can. 

shoot me your email when you get a chance. richn3@mac.com