pi3d/pi3d_demos #66: Problem after power cut

Duzeper, 2022-11-26:
I have had PictureFrame2020 running successfully and, pretty much non-stop, for nearly 2 years now (my word, doesn't time fly), but a problem has developed after a power cut.
I have rebooted the Pi and restarted PictureFrame 2020 and its start/stop timers and everything is working fine except that, at seemingly random times (I have not discerned any pattern to the occurrences) the next slide will not display - the display goes to a blank screen. Then, once again apparently randomly, slides will display normally.
This is clearly a display issue as if, using MQTT, I select 'Next Slide' or 'Previous Slide', then it will display normally. Also 'Text Refresh' will display the slide properly. In addition, text (Filename, Date etc) displays normally, and correctly, even when the slide does not. Clearly, therefore, PictureFrame2020 thinks it is displaying the slide correctly.

As I said, up until the power cut which, by the way, occurred during the night when PictureFrame2020 was not running, it was working perfectly.

Any ideas?

Martin Dashper

paddywwoof, 2022-11-26:
Hi, it does sound odd. Powering off without a proper system shutdown is supposed to potentially cause some kind of damage in the file system (or something OS related) but I've not experienced it. Occasional blank pictures was one symptom of pictures being too large so maybe there's been a corruption of something keeping track of memory capacity???I will have a Google later but the only thing I would suggest trying is to make a new SD card with a fresh OS and copy the app and images onto that and see if the problem persists.Paddy

Duzeper, 2022-11-26:
Thanks Paddy. I was not sure if this thread was still being monitored.
I note that I am running Pi OS V10 (Buster) but I see that there is a V11 (Bullseye) and I wonder if I should install the latest version anyway with:

sudo apt update
sudo apt full-upgrade

Is that correct and would that throw up any other issues?

PS
I might try to install the Pi OS in a virtual machine on my Windows PC in order to try out PicFrame.

Duzeper, 2022-11-26:
A quick update. This morning the text display is not working properly. The grey text bar appears but the date or filename shows a black bar with no text.

paddywwoof, 2022-11-26:
On the RPi there has b een a bit of a divergence with the move to bullseye and the use of the old (bcm) graphics drivers. i.e. the newer OS doesn't support them. The replacements are better in some respects but sometimes slower. If you look at the discussions [here](https://github.com/helgeerbe/picframe/discussions/257) and [here](https://github.com/helgeerbe/picframe/issues/195#issuecomment-982999227) you will see various ideas about how to make bullseye work better. It depends very much on the power of the RPi you are using.

I would be interested to hear how you get on with pi3d in a virtual machine on windows. That's on my list of things to check now that, as I understand it, WSL has graphics capabilities.

Duzeper, 2022-11-26:
I cannot find the original detailed instructions for installing Pi3D and PictureFrame2020. Can you remind me where to find them

Martin

Duzeper, 2022-11-26:
A quick update. I have replaced the image files on the Pi with the originals and that, so far,  seems to have fixed the issue.
Clearly, a long term test is required, but so far, so good.

I would still like to download the PictureFrame2020 instructions again but the link I have now points to PicFrame instructions.

Martin

paddywwoof, 2022-11-26:
Martin, Yes Wolfgang migrated to picframe as we had tweaked the code so much with various enhancements!

I will set up a new SD for one of my RPis here and check what needs to be done. But essentially I think:
1. setup OS as per normal instructions on raspberrypi.com, https://pi3d.github.io/html/ and the links above for bullseye.
2. `python3 -m pip install pi3d` for pi3d (on WSL or linux on laptop you probably need to `sudo git install mesa-utils-extra`)
3. Either clone the whole github.com/pi3d/pi3d_demos repo or just right click and save the four `PictureFrame2020....py` files into an appropriate directory (for instance `/home/pi/` if this setup is only to run the picture frame). If you don't clone the whole demo repo you will need to also save the font from `pi3d_demos/fonts/NotoSans-Regular.ttf` and shader from `pi3d_demos/shaders/blend_new` then change the relevant lines in the PictureFrame2020config.py file
4. If you're going to use geo info you need to install geopy and there might be other dependencies
5. put all your pictures into another directory (for instance `/home/pi/Pictures`)
6. edit `/home/pi/PictureFrame2020config.py` so the appropriate paths point to the right places (pictures, font, shader, missing img and geo chache txt file.
7. make sure it runs ok from the command line `python3 PictureFrame2020.py` and when it does set up systemd to run it as a service from scratch see https://github.com/helgeerbe/picframe/wiki/Running%20As%20Service

As I say, I haven't done this for a while so there might be errors and omissions - I will check back later!

Paddy

rec, 2022-11-27:
Wow! Did you keep the original files? It would be fascinating to take a
look at them, perhaps.

Duzeper, 2022-11-27:
I'm afraid not. I just overwrote the files on the Pi.
I think, however, that I was a bit premature. Although it had been working properly all day yesterday since I restored the image files (after starting the frame manually), when the frame started this morning it was back to not displaying images. I have now stopped the frame and restarted it manually and it is back to working properly (so far, at least).

I am using a frameOn.service, frameOn.timer and a frameOff.service, frameOff.timer, all of which ahd been working continuously for the last 2 years. It would seem that the powercut has caused an issue with one or more of these or, I suspect more likely, the system behind these processes (forgive my clunky terminology which will confirm that I am not a Linux person, so pretty much everything I learned 2 years ago I am having to relearn)
Anyway, I will need to do some more testing by resetting the times in the timers and rerunning them to confirm this.

I believe it is possible to do an OS update in situ, but, assuming there is a 'system' problem, is there also a 'repair' option?

Martin

PS. I have successfully installed 32bit & 64bit Buster in VirtualBox on my PC, but I would like to try and get PictureFrame2020 working again before I have a go at a virtual picFrame.
By the way, I have installed Buster, as Wolfgang seems to suggest that Bulldog may not be suitable for picFrame.
Also since I am running a Pi3B+, am I right in assuming that the 32bit OS is the correct one for me?

paddywwoof, 2022-11-27:
Martin, I have just now followed @tomasedoff instructions https://gist.github.com/thomasedoff/3973846101da1e7496524be075f1c02f pretty much exactly but instead of pip installing picframe, just installed pi3d and git cloned pi3d_demos. Transitions ran very slowly but after going into raspi-config and switching on advanced options -> glamor it seemed fine. Ctrl-C didn't stop it (I was running it with keyboard and mouse) so you need to be able to SSH in and kill the process while you test it. I will try putting the SD card into an old RPi and see what it looks like.

I'm not sure what caused your bug but it might be something corrupted on the SD card so it would make sense to try another.

Duzeper, 2022-11-30:
Paddy
I have a 'useful.txt' file containing what I thought might be some (believe it or not!) useful commands, and in it is:

'sudo pip3 install pi3d --upgrade'

Is this what I am looking for to update TextBlock.py?

Since I do not know what that file does, I am not sure if is causing an issue that I haven't yet spotted. Otherwise it all now seems to be working OK

Martin

paddywwoof, 2022-11-30:
Martin, sorry not reply before, I'm away from home and can't check stuff easily. The TextBlock.py file is part of the 'standard' pi3d so that would normally live in wherever pi3d is installed, inside the pi3d/util directory. So updating pi3d to the latest using pip upgrade should fix that in theory. There were some fixes 18 months ago so maybe you had a version in your PictureFrame2020.py folder to 'trick' the module loading!!

For picframe we moved from the TextBlock system using OpenGL point rendering because a) the fast text changing ability wasn't needed b) it was useful to be able to draw any character in the available font (i.e. thousands of unicode characters used in place names of the geopy system).

Keep an eye on your system and shout out if you have any other problems.

Paddy

Duzeper, 2022-11-30:
Thanks Paddy. I just wanted to check before I update and create a new updated clone. Fortunately the previous clone was not way out of date and everything seems to be back to normal.
I would really like to try out picFrame and, if I could actually buy another RPi, I certainly would. In the meantime I'll have a go on the virtual machine, though it's never quite the same is it.

Thanks for your help.
Martin

Duzeper, 2022-11-30:
Sorry, one more, general, question.
If I want to update my system, do I need to run both:

sudo apt update & sudo apt upgrade

In other words, does upgrade just upgrade my current OS (buster) or will it upgrade to the next version (bullseye), which I do not want to do?

Martin

paddywwoof, 2022-11-30:
I'm pretty sure it will stay with buster. I think there is a way to move on to the next version but the standard `update`, `upgrade` won't do that. As I understand it `update` sorts out the database of latest versions and all the dependencies between everything installed through the package manager. `upgrade` downloads and installs all the packages to match. (probably a *massive* over simplification.

Duzeper, 2022-11-30:
Many thanks.
Over simplification is my comfort zone.

Duzeper, 2022-12-07:
Just a quick note about running a virtual RPi.
I'm not sure it will work. Since the x86 architecture cannot run Raspbian, the only Raspberry Pi OS Desktop ISO files are actually Debian, incorporating the RPi desktop.
I have a working virtual machine, however, when I try to install picframe, following @tomasedoff instructions, there are various problems:

2. Connect to Wi-Fi network
```
sudo raspi-config nonint do_wifi_country SE               # Set the country code
sudo raspi-config nonint do_wifi_ssid_passphrase SSID PSK # Set ssid and passphrase 
```
```
sudo: raspi-config: command not found
```
However as virtual machine - Not Needed

3. Update Raspberry Pi OS
```
sudo apt-get update
sudo apt-get upgrade
```
No errors

4. Enable remote administration
```
sudo systemctl start ssh
sudo systemctl enable ssh
```
```
Failed to start ssh.service: Unit ssh.service not found
```
Again, not needed for virtual machine

5. Configure Raspberry Pi OS
```
sudo raspi-config nonint do_boot_behaviour B2                  # Enable automatic login
sudo raspi-config nonint do_blanking 1                         # Disable Xorg screen blanking

sudo raspi-config nonint do_change_timezone "Europe/Stockholm" # Set time zone
sudo raspi-config nonint do_hostname picframe                  # Set hostname
```
```
sudo: raspi-config: command not found
```
6. Install necessary software and dependencies
```
sudo apt-get install --no-install-recommends \
  xserver-xorg \
  xserver-xorg-legacy \
  xinit \
  python3-pip \
  libopenjp2-7 \
  libgles-dev \
  libatlas3-base \
  libxrender-dev
```
OK except 
```
E: Unable to locate package libgles-dev
```
7. Configure Xorg
```
sudo sed -i 's/\(^allowed_users=\).*/\1anybody/' /etc/X11/Xwrapper.config
sudo bash -c 'echo "needs_root_rights=yes" >> /etc/X11/Xwrapper.config'
```
No errors

8. Install and initialize PictureFrame
```
python3 -m pip install picframe
```
```
/usr/bin/python3: No module named pip
```
Clearly, at this point I cannot go any further.

Is it possible to rectify these issues, given that the virtual machine is actually a Debian installation, or am I fighting a losing battle?
Given that there seems to be no likelihood of being able to buy another RPi any time soon, is it possible to install both picframe and pictureframe2020 on the same machine (obviously not running them both) or will (as I suspect) the one interfere with the other?

All that aside, I can confirm that I now have PictureFrame2020 back working properly (and an up to date OS clone!)

Martin

rec, 2022-12-07:
>    1. Connect to Wi-Fi network
>    1. Update Raspberry Pi OS
>    1. Enable remote administration
>    1. Configure Raspberry Pi OS
>    1. Install necessary software and dependencies
>    1. Configure Xorg
>    1. Install and initialize PictureFrame

Man, sorry this has been so painful for you.

That last error is wack.  How can you have a python3 and no pip module?

It sounds like something is badly wrong: I dunno if this will help but try
these to at least get a hint of the issue.

python3 --version
which python3
which pip3

helgeerbe, 2022-12-07:
Hi @Duzeper 
_Is it possible to rectify these issues, given that the virtual machine is actually a Debian installation, or am I fighting a losing battle?_

I would say it depends, on what you are targeting. As far as I know there is is only one hardware emulation of the raspberry pi available. It's QEMU. But it doesn't support graphic rendering well.

But good news is, that picframe runs on linux if OpenGL and an X server is available. So I tested it on ubuntu, debian and actual developing picframe on MacOs. But you can't use raspi config tools, and sometimes some pathes to files are little different.

A good starting point is to install an actual debian linux with an X environment. Ensure that python and pip is installed. Install picframe `pip install picframe` and start it from a terminal window. Chances are quite good, that picframe will work. SSH and autostart of picframe is highly dependent on your running environment. But there you should use the standards of your linux distribution.

Duzeper, 2022-12-07:
Hi Tom
These are the results:
```
pi@RaspBerryPi:~$ python3 --version
Python 3.7.3

pi@RaspBerryPi:~$ which python3
/usr/bin/python3

pi@RaspBerryPi:~$ which pip3
pi@RaspBerryPi:~$ 
```
Martin

Duzeper, 2022-12-07:
Hi @helgeerbe 
I'm sorry but I am not a 'Linux person' so I am not sure what is meant by 'X environment'
I have found 'debian-10.12.0-i386-xfce-CD-1.iso ' on the Debian.org website. Is this suitable?
I am thinking v10 as I understand that there are issues with picframe and V11, and 32bit (i386) since I run an RPi 3B+ which I understand is 32bit. 

Martin

paddywwoof, 2022-12-07:
Martin, sorry not to get back earlier. I'm out and about at the moment but will reply more comprehensively later this afternoon.I haven't run PictureFrame in windows Linux but I have set it up recently in normal Ubuntu so I will let you know what I did for that. Much simpler than setup on RPi but with some deficiencies.Paddy

Duzeper, 2022-12-07:
I'm thinking now that I might give up on the Debian/RPi virtual machine and try @helgeerbe's suggestion of installing a straight Debian linux.
Since all I want to do is try out picframe, it clearly doesn't matter which version of linux I use in a virtual machine, so if I can get it working in Debian (or Ubuntu) then that is fine.

Martin

Martin

Duzeper, 2022-12-08:
OK, just installed Debian10 and gone through the picframe instructions again and got all the same errors as before. Clearly there is an issue with Debian.
I might have a go with Ubuntu tomorrow.

Martin

paddywwoof, 2022-12-08:
Martin, better hold on a while. There looks to be a problem with the heif dependency https://github.com/helgeerbe/picframe/issues/297
This is something @helgeerbe has just been working on.

I did try pi3d in various flavours of  Linux in a VM several years ago, Ubuntu and lubuntu we're fine but I did have problems with vanilla Debian. Of course I can't remember the details!

I will also try installing picframe on my laptop. I had it there previously but I've wiped and reinstalled the OS since.

Paddy

paddywwoof, 2022-12-08:
Martin, not tried picframe on this laptop yet but there does seem to be an odd issue installing the heif module caused by an older version of pip (maybe this kind of thing is why it's not part of python?). People seem to have fixed it by upgrading pip
``` console
$ python3 -m pip install --upgrade pip
```
You could try that and see if it fixes the errors...
Paddy

Duzeper, 2022-12-08:
Thanks,  That all appeared to work:
```
pi@RaspBerryPi:~$ sudo pip -V
pip 22.3.1 from /usr/local/lib/python3.7/dist-packages/pip (python 3.7)
```
Then I tried to install picframe ``` python3 -m pip install picframe ``` with no apparent errors:
 until at the end I get some warnings:
```
Installing collected packages: paho-mqtt, IPTCInfo3, PyYAML, pycparser, Pillow, numpy, defusedxml, pi3d, ninepatch, cffi, pi-heif, picframe
  WARNING: The scripts f2py, f2py3 and f2py3.7 are installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script picframe is installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed IPTCInfo3-2.1.4 Pillow-9.3.0 PyYAML-6.0 cffi-1.15.1 defusedxml-0.7.1 ninepatch-0.1.22 numpy-1.21.6 paho-mqtt-1.6.1 pi-heif-0.8.0 pi3d-2.49 picframe-2022.12.4 pycparser-2.21
```
Then after successfully completing Section 8, I move on to Section9. Test PictureFrame
```
pi@RaspBerryPi:~$ xinit /usr/bin/python3 /home/pi/.local/bin/picframe /home/pi/picframe_data/config/configuration.yaml

(EE) 
Fatal server error:
(EE) Server is already active for display 0
	If this server is no longer running, remove /tmp/.X0-lock
	and start again.
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) 
pi@RaspBerryPi:~$ XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
      after 7 requests (7 known processed) with 0 events remaining.
```
So, progress at least, but not quite there.

Martin

helgeerbe, 2022-12-08:
This looks good.  I ment to run it as separate commands.

Have a look https://github.com/helgeerbe/picframe/wiki/Command%20Line and https://github.com/helgeerbe/picframe/wiki/Configuration

Duzeper, 2022-12-20:
OK, I have, for the time being, put the virtual installation on hold since I have actually managed to obtain another RPi3B+, so I have started to install picframe onto this following Wolfgang's:
https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/

I have got as far as testing the installation:
```
picframe ~/picframe_data/config/configuration.yaml
```
Which results in:
```
pi@raspberrypi:~ $ picframe ~/picframe_data/config/configuration.yaml
INFO:start.py:starting ['/home/pi/.local/bin/picframe', '/home/pi/picframe_data/config/configuration.yaml']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
libEGL warning: DRI2: failed to authenticate
```
which needs a “CTRL+C” to exit

paddywwoof, 2022-12-20:
Martin, that looks like an error related to the OpenGL driver (libEGL). Have a look at the settings in `$ sudo raspi-config` and make sure they match the instructions in the RPi3 rather than RPi4 section. Unfortunately things are in a bit of flux at the moment with heif taking a while to settle down!

Duzeper, 2022-12-20:
All done as per RPi3. No errors but did get a warning:
```
pi@raspberrypi:~ $ sudo python3 -m pip install -U pip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /usr/local/lib/python3.9/dist-packages (22.3.1)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
```
Same warning after:
```
sudo python3 -m pip install pi-heif
```
Still get: ```libEGL warning: DRI2: failed to authenticate``` and no image displayed

paddywwoof, 2022-12-20:
I think that warning isn't what's causing your error (pip does say that it's already installed, and if it wasn't the rest wouldn't work).

Are you logged in via ssh with no desktop running? What happens if you open a python3 terminal then
```python
>>> import pi3d
>>> pi3d.PLATFORM
>>> d = pi3d.Display.create(w=100, h=100)
```
The old broadcom driver is `pi3d.PLATFORM_PI` which is `0` as opposed to the new mesa one (which is `3`). I would expect the Dispslay.create() to be the line causing the error but if it runs OK try adding
```python
>>> while d.loop_running():
...   pass
... 
```
and see if it's arising later. If you do `$ cat /etc/os-release` presumably you get "buster".

You could also try running the `$ sudo xinit ...` version of the command line test used for the RPi4 setup.

paddywwoof, 2022-12-20:
PS you could also check the driver being used with
```python
>>> import pi3d
>>> pi3d.openegl
```
which should give a DLL object from `/opt/vc/lib/libbrcmEGL.so`

Duzeper, 2022-12-20:
OK, logged in via ssh but desktop running:
```pi@raspberrypi:~ $ python3

pi@raspberrypi:~ $ python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pi3d
>>> pi3d.PLATFORM
0
>>> d = pi3d.Display.create(w=100, h=100)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.9/site-packages/pi3d/Display.py", line 569, in create
    display.opengl.create_display(x, y, w, h, depth=depth, samples=samples, layer=layer,
  File "/home/pi/.local/lib/python3.9/site-packages/pi3d/util/DisplayOpenGL.py", line 120, in create_display
    assert self.context != EGL_NO_CONTEXT and self.context is not None
AttributeError: 'DisplayOpenGL' object has no attribute 'context'
>>> pi3d.openegl
<CDLL 'libEGL.so.1', handle 2a4c36e0 at 0x7f82078580>
>>>

pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
I now have a recollection that Wolfgang (I think) had suggested that there may be graphics issues with 'Bullseye'. Is that the problem?
Should I start again but install 'Buster'?

helgeerbe, 2022-12-20:
I run it on bulleseye myself. But perfomace on buster is  actually better. On bulleseye the maintainer try to get rid of special rasbian code. But bulleseye is fine for me on a 3B+ with a resolution of 1920x1200.

Bulleseye need other settings: https://github.com/helgeerbe/picframe/discussions/257#discussioncomment-3689273

Duzeper, 2022-12-21:
Thanks. I may go with buster, if only because all of my limited experience is with that version. It may have to wait til after Christmas now.