pi3d/pi3d_demos #67: PictureFrame2020.py: ValueError: operands could not be broadcast together with shapes (200,3) (0,)

Home-Control9000, 2023-06-20:
Hi,

Im installing a new Ubuntu system for using as a photoframe. I have 3 old systems with no problems.
When im run  PictureFrame2020.py i get a following error.  

> python3 PictureFrame2020.py
Traceback (most recent call last):
  File "/home/walter/pi3d_demos/PictureFrame2020.py", line 498, in <module>
    text = pi3d.PointText(font, CAMERA, max_chars=200, point_size=config.SHOW_TEXT_SZ)
  File "/usr/local/lib/python3.10/dist-packages/pi3d/util/PointText.py", line 51, in __init__
    self.text = Points(camera=camera, vertices=self.locations, normals=self.normals,
  File "/usr/local/lib/python3.10/dist-packages/pi3d/shape/Points.py", line 34, in __init__
    if normals == [] and tex_coords == []:
**ValueError: operands could not be broadcast together with shapes (200,3) (0,)**

joshstir, 2023-09-26:
I know this is a bit ago, but I just ran into this same issue.  The problem is that the shape of normals and tex_coords is 200,3 and the shape of an array is (0,) so this comparison fails.  It looks like it's just checking if the array is empty, which could be fixed by switching the code to:` if normals.size == 0  and tex_coords.size == 0:`

paddywwoof, 2023-09-26:
Hi, sorry this has languished. I will look at it now. @joshstir, did you try altering the line as you suggest and did it work?

Paddy

Home-Control9000, 2023-09-26:
> I know this is a bit ago, but I just ran into this same issue. The problem is that the shape of normals and tex_coords is 200,3 and the shape of an array is (0,) so this comparison fails. It looks [...]

Thanks for the solution. I solved it a few months ago by downgrading to a older Ubuntu distribution which is not that elegant but it was the only option to make it work. I will try your fix when installing a new system.

Walter

paddywwoof, 2023-09-26:
Hi, I've just change the source to ```    if len(normals) == 0 and len(tex_coords) == 0:``` as the ```Points``` class is used elsewhere being passed normal python lists. The lists get converted into numpy arrays in ```Buffer.__init__()``` which can cope with lists or np arrays. Previously python would compare np arrays with empty lists but gave a deprecation warning (which I hadn't noticed until I ran some tests in a terminal just now. It still works OK on ubuntu 22, which I'm using here, but will obviously fail soon)

I will push a new version up to github and pypy later today so you should be able to ```pip install --upgrade``

joshstir, 2023-09-26:
Sorry for my late reply, that did fix it, but you've already got it fixed.  😃. Thanks for taking a look at this!