pi3d/pi3d_demos #69: Stopped working properly after apt-get upgrade

Tsjippy, 2025-04-23:
I love this and have ued it now for 2 years but letely it does not work anymore.
It just shows a black screen with flickering white squares at the bottom.

any clue?

There are no errors on the command line running ( python3 /home/pi/pi3d_demos/PictureFrame2020.py ) and I run it on an up to date rpi3B

paddywwoof, 2025-04-23:
Hi @Tsjippy that does sound odd, but there are 'improvements' popping up to the OS window manager from time to time that do impact on graphics rendering. I will try `PictureFrame2020` on the Pi3 here, which has the latest OS etc. Here are some questions I would ask to debug it myself:
1. Did the problems start after an upgrade, either to the OS or pi3d, if so can you remember what changed.
2. What do you get running other demos (Minimal, Minimal2d, ForestWalk etc.)
3. One thing we've done with the `github.com/helgeerbe/picframe` project is to default to using SDL2 to wrap the lower level window stuff. To get that to work you need to `sudo apt install libsdl2-dev`, then in the python file where the `Display` instance is created https://github.com/pi3d/pi3d_demos/blob/e17eb58663bca2d20118811e5ff9d197a690a070/PictureFrame2020.py#L473-L475 add the argument `use_sdl2=True`
Paddy

Tsjippy, 2025-04-23:
Thanks for your reply!

Minimal works
Minimal_2d: blue flickeringscreen
ForestWalk.py does not work: black unnamed window

I have run `sudo apt install libsdl2-dev`
Changed it the code to 
`DISPLAY = pi3d.Display.create(x=config.DISPLAY_X, y=config.DISPLAY_Y,
              w=config.DISPLAY_W, h=config.DISPLAY_H, frames_per_second=config.FPS,
              display_config=pi3d.DISPLAY_CONFIG_HIDE_CURSOR, background=config.BACKGROUND, use_sdl2=True)`

But get:` TypeError: create() got an unexpected keyword argument 'use_sdl2'`

paddywwoof, 2025-04-23:
Ah, you must be using an older version of pi3d. There are probably some fixes to get round other changes to the RPi OS, not sure if and when changes were put in. You can update it with `python3 -m pip install pi3d --upgrade` assuming you're not using the `venv` virtual environment (which seems to be the case from the description of what you typed on the command line in your first post). There are some advantages to using venv and the python community is leaning that way, so things might increasingly become harder to get working without it.

Hopefully upgrading pi3d won't break anything else! Let me know how you get on.

closed 2025-04-23 by Tsjippy